system76/ec #3: Discussion: potentiall EC debugging/development tool

mokus0, 2019-12-15:
I'm not sure if this is the right place for this discussion, or if there is a better one, but I'm interested in trying a bit of a different approach to EC development and would like to gauge interest.

In particular, I'd like to develop a replacement for the ITE8587 in my galp4, probably based on the STM32L4R5 or similar, for a few reasons:
* I am pretty familiar with that MCU, and I think I may even be able to implement LPC using its DMA engine - otherwise, I'd pair it with a very small ICE40 FPGA for LPC support
* It's a pretty powerful MCU, but still very low power.  I am also, in general, interested in any power optimization I can do to my system especially in suspend states (looking over the schematic, there are a lot of "freebie" improvements I have spotted already).  Power optimization is a big part of what I do at work.
* There is a part from Maxim (MAX6621) that can be used to implement PECI.

As a first step toward understanding more of the low-level messaging of the LPC bus, I've started working on a dev tool in the form of a m.2 3042 module that can be installed in the 3G module slot and includes an ICE40 FPGA in a configuration that should be compatible with the ICEstick dev tools, as well as the MAX6621 IC to experiment a bit with.

Is there any general interest in this concept, or even just in the m.2 FPGA board for dev use?  My time is rather limited so I don't know how consistently I can work on it but if there is interested I'd be more motivated to continue on it.  By some very rough estimates, I think I could probably fabricate 20 of these boards for under $50/ea, less if people are comfortable hand-assembling them.

mokus0, 2019-12-15:
The hardware design is hosted here: https://github.com/mokus0/m2-ice40
The schematic is nearly complete for a first draft, but still needs review and FPGA I/O pin assignments, test points, etc.

jackpot51, 2019-12-16:
The only place where LPC is accessible on the galp4 motherboard is where the IT8587E is soldered. It would have to be removed and replaced with a pin compatible replacement. I find it unlikely that this could be done with another chip.

mokus0, 2019-12-16:
> The only place where LPC is accessible on the galp4 motherboard is where the IT8587E is soldered. It would have to be removed and replaced with a pin compatible replacement. I find it unlikely that this could be done with another chip.

My plan is to design a small PCB carrying the replacement circuit and solder it in place of the original EC. I am very comfortable with this level of rework.

closed 2024-01-02 by crawfxrd