system76/ec #9: Study fan curve

WatchMkr, 2019-12-16:


jackpot51, 2020-04-03:
**Problems**

Things I am annoyed by:
1. fan running at all
2. fan changing speed frequently
3. system throttling
4. system being hot to the touch

Balancing these things is very important. The user expects the fans to turn on when doing heavy workloads. The user expects the system to be silent under light workloads. The user expects the ramp up and ramp down to be gradual.

Unfortunately for the user, the ramp up and ramp down of mobile CPUs is anything but gradual. Going from 1W to 25W in a split second and having very small heatsinks means the fan needs to turn on quickly to avoid thermal throttling.

Additionally, heat builds up on the heatsink and chassis and this heat if not extracted from the system can keep the CPU temperature hot even when it goes back from 25W to 1W. This residual heat can also cause the temperature jumps to happen right around the fan cutoff temperature, making the fan speed spikes even larger.

For example, the current fan turn-on point is 65 degrees. When you break that temperature, once the CPU is idle, it will drop below 65 degrees. This will then turn off the fan. The CPU will very slowly drop in temperature with the fan off, so a jump in power output from the CPU can boost temperatures from 65 degrees sometimes to the CPU's throttling point, causing the CPU fan to literally go from 0 to 100.

Contrast this with the behaviour when the system has been idle for a long time, with temperatures around 35 degrees. It takes similar power output to bring the CPU from 35 degrees to 65 degrees so a CPU power output jump that would bring an already hot system from a fan speed of 0 to a fan speed of 100 would cause absolutely no change in fan speed if the temperature rises to just below the turn-on point.

So, what are the problems with the current fan curve? In attempting to minimize how much the fan runs, point number 1 and 3, we have maximized points number 2 and 4. The system will now run close to the fan turn-on point often during moderate use. Sure, it will cool effectively when nearing the throttling point, and it will spin down often - but it will eventually reach uncomfortable temperatures and have an overshoot effect causing the fan to change speed all the time due to variable workloads.

**Solutions**

How can this be addressed? There are a few methods to keep the system cool and reduce the number of fan speed changes.

First, have a fan turn-on point that is higher than the fan turn-off point. If the fan turns on at 65 degrees, it could turn off at 50 degrees, for example. There would be a curve for increasing the fan speed, and a curve for decreasing the fan speed that was shifted lower. This ensures that the system cools down enough that small jumps in CPU energy output do not require the fans, and that the system remains comfortable to the touch.

Second, smooth out changes to fan speeds. When the system is under load, it is common for the load to be variable and cause CPU power output to vary drastically. This is often caused by having bottlenecks other than the CPU. In these cases, the current fan curve would jump around, following the CPU power output. Having consistent fan speeds during such a workflow is critical to improving user experience. One way to do this is to introduce a timeout parameter, increasing based on the CPU temperature. This would keep the fans spinning based on the highest temperature the CPU had reached even after the CPU lowered in temperature.

**TLDR**

With the following changes, we get closer to modelling the ideal thermal solution, based on our limitations of just having CPU temperature as an input and fan speed as an output:
- Have a second curve for decreasing fan speed
- Have a timeout for allowing the measured temperature to drop

jackpot51, 2020-04-05:
Solved by #43. Please create additional issues for improvement.

closed 2020-04-05 by jackpot51