system76/ec #10: ACPI error leads to HiDPI daemon rescaling screen

brs17, 2019-12-20:
On the galp4 and darp6 running this EC, after logging in, after 30 seconds or so, the scaling on the display goes from 100% to 200%. This does not happen if the HiDPI daemon is off, or on a darp6 without the open EC. 

The error returned via dmesg:

```ACPI Error: Aborting method \_SB.LID0._LID due to previous error (AE_TIME) (20190703/psparse-531)```

jackpot51, 2020-02-03:
Merged with #12

jackpot51, 2020-03-31:
I haven't seen this in a while, has anyone else? @brs17 @crawfxrd  @leviport @WatchMkr

crawfxrd, 2020-03-31:
I still see this on the galp3-c.

jackpot51, 2020-03-31:
Does it happen without parallel debug?

crawfxrd, 2020-03-31:
Yes. On d96147688d6e7 with all debugging disabled I still get these messages.

jackpot51, 2020-03-31:
Interesting

brs17, 2020-04-01:
I don't think I had seen this either recently on a galp4.

jackpot51, 2020-04-01:
Will close until it comes back.

closed 2020-04-01 by jackpot51