system76/ec #23: darp6: System fails second suspend

crawfxrd, 2020-02-20:
Version: [2020-02-18_f7453df](https://github.com/system76/ec/commit/f7453df)
Target: `system76/darp5`

**Steps to reproduce**:
- Press `Fn`+`F12` to sleep
- Power Power button to wake
- Press `Fn`+`F12` to sleep

**Expected behavior**:
- screen turns off
- power button turns off
- power status LED blinks green

**Actual behavior**:
- screen turns off
- power button is _on_
- power status LED is _solid_ green

brs17, 2020-02-20:
Did you find you needed to be unplugged from AC power to replicate this?

brs17, 2020-02-20:
This issue was also replicated on the galp4 and darp6.

crawfxrd, 2020-02-20:
No, I'm AC power.

`ecflash/ec.py` is also failing to read the EC version on the darp6.

jackpot51, 2020-02-20:
You guys have open and proprietary EC mixed up. @brs17 saw this behaviour on proprietary EC. @crawfxrd saw this behaviour on open EC. The ec.py script is not adapted for the open EC yet.

jackpot51, 2020-02-20:
This could point to it being a coreboot issue

crawfxrd, 2020-03-13:
Closing in favor of #31

closed 2020-03-13 by crawfxrd