system76/ec #37: Darp5: Battery drains a bit while machine is off

leviport, 2020-03-09:
I've noticed that while my darps 5 are off, they still slowly consume some battery. It isn't much, but it is evident over a few days. For example, I had my work Darter plugged in and fully charged on Friday before I left. I turned it off and stuck it in my backpack, where it stayed all weekend. When I fired it up this morning (Monday), the battery was at 72%. I'm wondering if this is related to https://github.com/system76/ec/issues/34 and something just remains running in a low power state after shutdown.

jackpot51, 2020-03-10:
It probably is related

brs17, 2020-03-12:
Similarly, I experienced odd behavior on galp4 with open ec when the system shutdown on low battery. The status lights remained on and I had to hold the power button to shut the system completely down. Once booting it back up, the system did not consistently charge (sitting at 18% for quite a while). Once I noticed this, I turned the system off, unplugged the AC adapter and then once I turned it back on, the machine charged as expected.

crawfxrd, 2020-03-12:
I left the galp3-c powered off and unplugged last night. When I started it this morning, it was at 76%.

jackpot51, 2020-03-17:
This may be fixed with the latest version. Please retest

crawfxrd, 2020-03-18:
Left powered off and unplugged for the night. Started this morning with the battery at 95%

jackpot51, 2020-03-18:
Fixed by https://github.com/system76/ec/commit/7fbad82088ff534bbe5495d2080889dcebeb5788

closed 2020-03-18 by jackpot51