system76/ec #38: Darp5: Fans slow down after CPU thermal throttles

leviport, 2020-03-17:
If I run a stress test, I notice that when the CPU thermal throttles and the temp drops a little, the fans slow down and stay at that speed. It seems like that keeps it in the throttled state since it doesn't cool down enough to un-throttle.

MilesBHuff, 2021-04-20:
Same problem on oryp7.

leviport, 2021-04-20:
There is no "System76 default fan curve". They are different for all models, and a lot of effort and tuning goes into them. Throttling temps are reached within the CPU way sooner than the heat has a chance to be soaked up into the heatsink. This leads to brief high temp blips, which quickly soak into the heatsink, lowering the temp back down to normal. Sustained loads hold the temp up higher for longer, which is when the fan _should_ be spinning up, in order to help cool the heatsink. If you spin the fan up to max for every high temp blip, you will have an incredibly annoying laptop. Remember: the fan cools the heatsink and the heatsink cools the CPU. 

Also, this is definitely not the same problem. Open EC for darp5 was in early development when I opened this issue, and I simply forgot to close it when it was addressed. Since it is no longer a problem on darp5 and a discussion is likely to spawn in https://github.com/system76/ec/pull/179, I'm going to close this issue.

closed 2021-04-20 by leviport

MilesBHuff, 2021-04-20:
@leviport Right, of course.  And that makes sense, and is good.  But the problem I was talking about, is when running the default fan curve on the oryp7 and then stressing 8 cores, the CPU goes up to 90'C for a while (10s? 20s?  I haven't timed it.), then down to 70'C for a while (about as long), then back to 90'C, etc.  This is without a change in system load.