system76/ec #39: USB Audio Devices Only Play System Sounds

WatchMkr, 2020-03-25:
When selected out the audio output device, USB audio devices play system sounds like beeps only. All other sounds go through the internal laptops speakers.

galp4 w/ open ec
@brs17 let's check this on the lemp9 as well

jackpot51, 2020-03-26:
This needs to be verified on 19.10 on either the darp6, galp4, or lemp9.

jackpot51, 2020-03-26:
On the same lemp9, I have 18.04 and 20.04 installed. On 18.04, USB-C headphones work and will play all sounds. On 20.04, USB-C headphones will only play system sounds.

This is more likely a 20.04 issue. I recommend testing 20.04 on the lab galp4 with the proprietary EC to rule out open EC.

jackpot51, 2020-03-26:
Additionally, my bluetooth headphones have the same issue. They will play system sounds but never all audio.

jackpot51, 2020-03-26:
Moved to https://github.com/pop-os/pop/issues/886

closed 2020-03-26 by jackpot51