system76/ec #42: lemp9: Left side of system heats up when AC adapter is plugged in

brs17, 2020-04-02:
The A/C barrel plug adapter for the lemp9 is on the left side of the system. When the adapter is plugged in the left side of the system gets warm, if not hot, to the touch.

At first I just thought it was the NVMe drive installed on the left side of the system, but then realized that the left third of the chassis as a whole gets warm.

jackpot51, 2020-04-05:
Solved by https://github.com/system76/ec/commit/ed0e662d60dfb19951cca8384326eca5fc117fbc

Charging current was halved. System charges at 30W instead of the previous 60W and should not heat up as much while charging.

blakeridgway, 2020-06-04:
I have been noticing a similar issue, this issue was brought to my attention. 

When the battery gets to about 80% charge the chassis gets noticeably warm with the barrel connector plugged in. According to the Vitals Temperature monitor my Lemur Pro starts sitting around 66C, and once the barrel connector is unplugged it will instantly drop down 10C.

OS: 
契cat /etc/os-release
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os

Firmware: System76 Lemur Pro (lemp9) 2020-05-15_e2baafa.

priyasiddharth, 2020-08-24:
I still see this issue with `fw 2020_08_6_2e5f452`

```
cat /etc/os-release
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os
```
Following https://github.com/system76/ec/issues/42#issuecomment-609114940, can the charging power be reduced further e.g. 20W? Is there a limit to which it can be reduced?