system76/ec #45: galp4 - Disk Decrypt Screen Doesn't Show External Display

WatchMkr, 2020-04-05:
Plugged in via USB Type C

@system76/quality-assurance please check the lemp9 and other display ports.

leviport, 2020-04-06:
I was testing this on my darp5, and I got the same behavior on the mini displayport and usb-c port, but I did see the lock screen when using HDMI.

brs17, 2020-04-07:
Neither of these behaviors sound out of the ordinary to me. In wanting to confirm that, I took a look at the oryp5. The oryp5 did not display the disk decryption screen on any external display (usb-c/display port, nor HDMI).

brs17, 2020-04-07:
I had no issues with usb-c nor hdmi on the lemp9 in regards to this issue.

WatchMkr, 2020-04-08:
Needs to be fixed before releasing open ec to galp3-c/galp4/darp6 customers.