system76/ec #55: Fn layer does not take into account previously pressed keys

meklort, 2020-04-21:
When pressing a Fn-enable key, such as ` (play/pause), one of two sequences can happen:
(1) Fn is pressed, ` is pressed, ` is released, Fn is release.
(2) Fn is pressed, ` is pressed, Fn is released, ` is released. This is how I normally type.

If case (1) above is used, everything behaves correctly.
If case (2) above is used instead, the following happens:
play/pase is sent to the OS.
` is sent to the os.

This is because there is a global layer selection instead of a per-key layer selection.
If a key is pressed (play/pause) when Fn is pressed, then that key should release as its fn key even after fn is released.
Conversely, If a key is pressed without fn depressed, and then fn is depressed after, I would also expect it to release as the non-fn key.

Effectively, this means that each key should track either what was depressed or the layer that is was on when depressed so that it can be released as the same key.

Let me know if you'd prefer that I work on a patch, however it may take me about a week before I can get to it.

meklort, 2020-04-21:
Also note that I have only tested this on lemp9.

jackpot51, 2020-04-21:
I have a fix for this in #56

meklort, 2020-04-21:
I've tried thee change out on my machine and it's now behaving how I expect. Thanks.

closed 2020-04-23 by jackpot51

jackpot51, 2020-04-23:
@meklort are you using a custom keyboard mapping? If so, I'd love to hear about it

meklort, 2020-04-23:
@jackpot51 Yes, I have setup a custom keymap.
You can find the version I'm using here: https://github.com/meklort/ec/blob/keymap/src/board/system76/lemp9/keymap/default.c

The main differences are:
- Enable Fn-lock for the F keys (default is the Fn-value instead of F1/F2/etc, Fn must be pressed to enable F1/F2/etc). In my normal use case, I'm much more likely to press volume up/down or brightness up/down than to use F1/F2/etc.
- Add a new Fn key for spacebar to match Thinkpad laptops. Fn+Spacebar toggles keyboard backlight. This is a much easier keypress to find when you are in a dark room than F4.
- Replace PgUp/PgDn with left/right when not in the Fn layer. I often accidentally press PgUp/PgDn when really meaning left/right, so I've effectively made this a large left/right key like on MacBooks. I have not yet decided if I'll keep this as-is or not.

There are a few other options that I'm thinking about that would repurpose some of the Fn keys for fast forward/rewind, but I haven't decide yet

WatchMkr, 2020-04-23:
FN+Spacebar is a nice combo for kb backlight.

PR #56 (2020-04-23): Fix release of key after release of layer key
This fixes #55