system76/ec #58: Ghost key detection blocking normal user input

ben-dyer, 2020-04-28:
Hello :wave: new system76 user here! I am experiencing the following issue on lemp9:

For a certain set of words, I misspell them essentially 100% of the time (e.g. "list" -> "lit" or ".org" -> ".or"), which is presumably due to my style of user input being incompatible with ghost key detection rules https://github.com/system76/ec/pull/40. 

I have seen #57 and it looks clear that this is an intended behaviour, but its disruptive so I am interested in an adjustment. Is it possible to configure the ghost key detection?

(:thinking: :thought_balloon: Is it as simple as cloning, modifying `src/board/system76/lemp9/kbscan.c` and flashing `make BOARD=system76/lemp9 flash_internal`? :sweat_smile: )

crawfxrd, 2020-04-28:
The current ghost key detection is as good as it gets, based on QMK's implementation. There is no configuration.

The file has moved to `src/board/system76/common/kbscan.c`. Your flash command is correct.

ben-dyer, 2020-04-29:
@crawfxrd Ok, thanks for the info!

jackpot51, 2020-04-29:
You may also want to look into debounce configuration. It is currently set to debounce after 20ms.

ben-dyer, 2020-04-30:
@jackpot51 I had a similar idea yesterday - was able to turn ghost key presses back on and used xev to check that most of the ghost presses occured in between 20-40 ms whereas normal presses were usually more than 60ms apart.

Turning debounce ms up to 40 removed some of the presses but for the words I have looked at it removes the wrong presses "list" is "liut" instead of "liust"  and build is "buisd" ("buisld" with 20 ms debounce).

Wondering whether a custom rule for what to do with a multi-key event based on a short history of keypresses might be possible. (Maybe it will turn out this is buggy, but then again maybe it would just work)

meklort, 2020-05-01:
@ben-dyer and @jackpot51 There's another possibility that may be worth looking at:
As mentioned, when you press and hold 'lis', the s key is rejected, as you mentioned. You'll also notice that if you *release* the 'l' and 'i' keys, the 's' key is *not* sent.

Looking at the code, this is due to the following:
```
            if (kbscan_has_ghost_in_row(i, new)) {
                kbscan_last[i] = new;
                continue;
            }
```
The kbscan_last[i] = new; line causes the s key to to be marked as pressed, and as such it will never be sent until it's released.
I expect a modification could be made such that on release of the l and i keys (possibly after a debounce), the ghost can be detected as removed, and then the s key could be sent.

Depending on your typing style, this may work, although I could see multiple keypresses being reordered incorrectly if they happen to be masked by the ghosting code at the same time.

jackpot51, 2020-05-01:
Good find, I'll see what I can do

meklort, 2020-05-02:
I did a first pass on this, and things seem improved with my very minimal testing. You can see the changes at meklort/ec@b9b101ae619648e885cc023558c353bc4aeb52c8

@ben-dyer You might give it a try and see if it helps. If the debounce is set wrong you will see both ghost values popping up ('lis' ghosts with 'u', and so you could see 'lius'). Note that this is based on top of another branch of mine with a customized keymap, so you'll probably just want to cherry-pick it.

ben-dyer, 2020-05-02:
@meklort Picked your commit, and all I can say is wow, this is working beautifully! Thank you so much for sharing!

Haven't tested for an extended time yet, but all of my problem words seem completely resolved, and I cannot detect any issue.

In case anyone is doing some QA, here is my short list of problem words which I observed before or after re-enabling ghost keypresses:

list (u)
build (s)
version (5)
upgrade (tab and semicolon)
org

partofthething, 2020-05-02:
Also picked @meklort's commit and flashed on new lemur pro. Work's great for all my problem words!

- builds
- once
- noise
- gives

All work great!!! So happy. This was the major thing I was not liking on my lemur.

closed 2020-05-04 by jackpot51

Ivshti, 2023-05-29:
@jackpot51 @meklort I am encountering this right now on a Lemur Pro.
For me, it's skipping the 'o' in words that end in sion like decision, version, extension. The issue is very timing related as it doesn't happen with slower typing. I have verified it on two separate Lemurs, both with Arch and PopOS.

It skips on "reason" too but I have to put a lot of effort into typing it quickly to get it to skip there.

Ivshti, 2023-05-29:
I did this as an experiment
```
 // Debounce time in milliseconds
-#define DEBOUNCE_DELAY 15
+#define DEBOUNCE_DELAY 1
 
 static uint8_t kbscan_get_row(uint8_t i) {
     // Report all keys as released when lid is closed
@@ -340,10 +340,10 @@ void kbscan_event(void) {
                 kbscan_ghost[i] = true;
                 continue;
             } else if (kbscan_ghost[i]) {
-                kbscan_ghost[i] = false;
+                // kbscan_ghost[i] = false;
                 // Debounce to allow remaining ghosts to settle.
-                debounce = true;
-                debounce_time = time_get();
+                //debounce = true;
+                //debounce_time = time_get();
             }
 
             // A key was pressed or released
```

Then flashed it and now I no longer get key skips. I do get the occasional ghost though... like for example there would be a 5 in rare occassions in "version".

Any ideas on how to fix it permanently?

ahoneybun, 2023-05-30:
This is a lemp11 not a lemp9 for the record.

crawfxrd, 2023-05-30:
> Any ideas on how to fix it permanently?

You don't. For firmware-open boards, only the bonw14 had NKRO. Everything else must use software debounce logic to avoid sending ghost keys.

Ivshti, 2023-05-31:
> For the debounce value, you have to compromise between not sending extra key presses and missing the key from hitting it too fast.

@crawfxrd point is, how is this value tested? Is there any scientific-ish way to estimate the correct value? For me, currently, it seems to be working amazingly with 3 ms. I tried 10, but it was still skipping, albeit less frequently.