system76/ec #59: rust error while flashing: #![feature]` may not be used on the beta release channel

partofthething, 2020-05-02:
I'm trying to flash to try out the recent keyboard fixes. Using both the release and beta channels of rust, I'm getting the error below. I understand that I can switch to the nightly channel of rust to build this but I have to say I'm uncomfortable flashing the system with a toolchain in the nightly status. 

```
   Compiling redox_hwio v0.1.1
error[E0554]: `#![feature]` may not be used on the beta release channel
 --> /home/nick/.cargo/registry/src/github.com-1ecc6299db9ec823/redox_hwio-0.1.1/src/lib.rs:2:93
  |
2 | #![cfg_attr(all(not(feature = "stable"), any(target_arch = "x86", target_arch = "x86_64")), feature(asm))]
  |                                                                                             ^^^^^^^^^^^^

error[E0554]: `#![feature]` may not be used on the beta release channel
 --> /home/nick/.cargo/registry/src/github.com-1ecc6299db9ec823/redox_hwio-0.1.1/src/lib.rs:3:38
  |
3 | #![cfg_attr(not(feature = "stable"), feature(const_fn))]
  |        
```

jackpot51, 2020-05-03:
You absolutely must use the nightly channel. Inline assembly is required for accessing the EC from Rust, and that is only available inthe nightly toolchain.

closed 2020-05-03 by jackpot51

partofthething, 2020-05-04:
Ok. Would you be interested in a PR updating the README to say that you have to have the nightly build of rust in the dependencies? That would have helped me a lot.

jackpot51, 2020-05-04:
Absolutely!