system76/ec #62: Battery charging threshold

tsitavets, 2020-05-09:
It would be great to implement battery charging threshold to increase battery lifespan when working a lot from AC. For example stop charging at 80% and start charging again at 60%. It would be nice to have keyboard shortcut to enable/disable feature and to have a possibility to change thresholds somehow (because optimal thresholds levels depend on battery usage). Also we can blink by power led to tell user is feature enabled (one time blink) /disabled (two times blink) upon pressing keyboard shortcut.

crawfxrd, 2020-09-11:
Battery charging thresholds are implemented. Currently compile-time only. Need to add ACPI support and expose it in sysfs.

crawfxrd, 2020-09-23:
Charging thresholds can now be set/queried with  `SBCT`/`GBCT` ACPI methods [\[1\]](https://github.com/system76/coreboot/pull/15).

`system76_acpi` has been updated to expose data in standard sysfs entries `/sys/class/power_supply/BAT0/charge_control_{start,end}_threshold` [\[2\]](https://github.com/pop-os/system76-acpi-dkms/pull/4).

See also: pop-os/gnome-control-center#101

closed 2020-09-23 by crawfxrd