system76/ec #69: lemp9: Can't resume work after suspend

tsitavets, 2020-06-04:
Sometimes laptop glitches after I opened it from suspend mode (put it to suspend mode just by closing lid): screen is black, power led is on and green, it responses only to changing of keyboard backlight but with strange blinking if I press several times in a row, laptop is a little warm and by closing the lid it's not going back to suspend mode (power led is not blinking). I tried to close/open lid again, to connect to external monitor, to press power button and all buttons/shortcuts on keyboard (including Fn+F2 and brightness changing) but nothing helped. The only way I found to fix this is to turn it off by long pressing power button. I had this issue already 2 times in last 2 weeks of not extensive usage of the laptop and I'm afraid to start loosing data when I actually start working on it.

jackpot51, 2020-06-04:
What is your firmware version, and what operating system are you running?

tsitavets, 2020-06-04:
**Firmware**: 2020-05-15_e2baafa
**OS**: Pop!_OS 20.04

I did clean installation yesterday (with wiping all data and repartitioning) and today in the morning I had the issue. Installed not too much software and some OS updates (after OS update did restart)

jackpot51, 2020-06-06:
Please create a support ticket, if you haven't already

tsitavets, 2020-06-09:
I'm closing the issue and will reopen if it's occur once again

closed 2020-06-09 by tsitavets