system76/ec #71: galp3 - Power-on when docked, without opening the lid

asahaf, 2020-06-10:
Currently, galp3 cannot be powered-on while the lid is closed. 
When I'm at the office, I would like to mount my galp3-c lid closed on a vertical stand (since I don't have space), hock up the dock station USB C cable and power it on, lid closed. 

I don't know if that's possible, but it should be possible to power it on while the lid is close when the power source is connected.

Thanks,

RemingtonHolder, 2020-12-30:
I feel like this could *probably* be somewhat easily implemented on the backend but would also take a deal of front end integration too, so I would also mention this on the [firmware repo](https://github.com/system76/firmware-open) too.

RemingtonHolder, 2020-12-30:
This is a prevention measure to make sure someone does not turn on their computer when not intending to. I would recommend posting this issue on the firmware repo too so they can integrate a switchable setting.