system76/ec #81: Actually disable lemp9 power button when off, AC disconnected, and lid closed

jackpot51, 2020-07-22:
This feature is currently only preventing power button presses when suspended, not when off.

RemingtonHolder, 2020-12-30:
Where does the power button signal get read to for starting the machine?

RemingtonHolder, 2020-12-30:
Isn't the power button handled in the common board code? Is this a problem with any other system?

RemingtonHolder, 2020-12-30:
This should be handled in the common board code here.
https://github.com/system76/ec/blob/42f1b4863fe5c41c76870acf4436ec3f349ae9be/src/board/system76/common/power.c#L393-L399

jackpot51, 2020-12-30:
There is some glitching due to the electrical design. This commit takes care of that:

https://github.com/system76/ec/pull/121/commits/954082178a093e46b62e7889747b42671801a6d0

RemingtonHolder, 2020-12-30:
@jackpot51  Is this a persisting issue with lemp10?

jackpot51, 2021-01-01:
lemp10 has the above firmware fix applied