system76/ec #103: Implement max fan keypress

jackpot51, 2020-09-21:
This used to be fn+1. It could be the start of a manual fan control framework

WatchMkr, 2020-09-21:
I recall regular support request from customers accidentally hitting the FN+1 combination.

Understanding how it might be used would help decide the best way to provide a max-fan feature (if at all).

jackpot51, 2020-09-21:
Support has stated that they prefer the small number of complaints about accidentally hitting it to being unable to test fans easily. It is also possible that having support for writing fan duty cycle from the OS would solve this problem, but is harder to implement

WatchMkr, 2020-09-21:
Got it. Let's use a three-key combo to allow testing and avoid accidentally turning on the fans to max. FN+Shift+1?

jackpot51, 2020-09-21:
I like two handed keypresses to avoid issues. Any thoughts on a right handed key that could be used?

WatchMkr, 2020-09-21:
Either side could be used with Shift in the combo. Keeping FN and 1 is good so that if it's not one (old stuff) add Shift (new stuff).

curiousercreative, 2021-01-06:
Was this issue resolved?

jackpot51, 2021-01-06:
Yes, it was resolved. Fn-1 triggers maximum fan speed

closed 2021-01-06 by jackpot51