system76/ec #112: lemp9 > USB-C DP and Charging Not Working from Cold Boot

WatchMkr, 2020-09-28:
While testing a display, I found that charging and DP did not work after cold booting with the monitor plugged in. After unplugging and plugging in the display, both work. I haven't tested other laptop models and only had the lemp9 for a short period so I couldn't confirm on other displays.

https://www.dropbox.com/s/auf4jags0snswld/lemp9-usbc.mp4?dl=0

leviport, 2020-09-28:
Hmm, I've not seen that on the lemp9 I've been using, and I've been using my external 4K display with a USB-C cable with it for a long time. My display does not charge over USB-C though, so I bet it's unique to that scenario.

RemingtonHolder, 2020-12-31:
I think this might be a problem with this specific ANX7411 USB-C port controller. From reading their documentation it seems as if the port can detect cable attach and detach and that is when the role direction setting takes place (as input, output, or both). It seems, with the way they do, when the port initializes it is given one of these two directional states, this being DFP(downstream facing port) or DRP(dual role port). Now, all the stuff I just said works fine when something is plugged in after the USB-C controller is initialized, but when something is already plugged in, the port does not set a transfer direction and instead sits in a floating state until the reattachment of the cable invokes the directional state to finally be set.

As far as @leviport situation would go, my best guess would be that the USB-C controller, because it was in a monodirectional state, initializes normally as it should.

This almost appears to be an unaccounted-for edge-case that the firmware on the USB-C controller does not know how the handle. Seems to me to be a massive oversight by Analogix.

Of course, this is only a best guess but (with the information I have *albeit this being a limited knowledge on USB-C controlers*) seems to fit what is happening.

cmm, 2025-01-22:
It reproduces consistently for me on lemp11.  Not a huge deal (not like I reboot every hour), but it definitely is annoying.

cmm, 2025-01-22:
@leviport "consistently" as in 100%.  What I have here is a Lenovo display which is also a charger, USB hub (several, in fact) and a KVM switch all-in-one, and the laptop senses none of the first three functions on boot (doesn't matter whether cold or hot BTW), until either the cable is disconnected/reconnected, or the display is power-cycled.

I forgot to mention that this specific display is annoyingly eager to enter power-saving mode (the feature is not configurable), so it is always in power-saving mode when the laptop boots up and the kernel (presumably) probes the device that the USB-C port sits under.  Might be an important detail, I guess?  Unfortunately I don't have other USB-C displays on hand to compare.

Also could maybe software be to blame, like something missing (or extraneous) in the initrd image?  But since reconnecting the cable brings the hub (and charger, probably)  to life while the LUKS prompt is still displayed (i.e. the inird system is still active), and after booting the real system the display wakes up too, I guess the software side works as well as it can.