system76/ec #125: make flash_internal fails - missing cargo (rust)

technicolorenvy, 2020-10-16:
great docs and examples, however it is worth mentioning that my newly shipped (late 2020) Oryx did not ship with rust installed so I was met with an error indicating that I was missing cargo. Googling about led to a simple solution - install rust (https://www.rust-lang.org/tools/install). 

It is probably worth adding this note somewhere to the docs.

crawfxrd, 2020-10-16:
System76 EC is part of the Open Firmware project. The full list of dependencies can be installed from [`deps.sh`](https://github.com/system76/firmware-open/blob/master/scripts/deps.sh).

technicolorenvy, 2020-10-16:
Oh, derp, literallly in the last section of the readme! Thanks and sorry for the noise!

closed 2020-10-16 by technicolorenvy