system76/ec #128: Charging thresholds don't persist through an EC reset

leviport, 2020-10-28:
If I set custom charging thresholds on oryp6, they reset when shutting the machine down and disconnecting the charger. I discovered this while testing https://github.com/pop-os/system76-power/pull/197.

crawfxrd, 2020-10-28:
Could be done with #63.

crawfxrd, 2024-08-14:
Using a minimized implementation of #256 for just battery thresholds:

- 4 bytes beginning at second to last sector (EC flash size - 2 KiB)
  - Avoids sector used for saving dynamic keyboard layout
    - Note: Hard-coded at 0x1FC00, even on 256 KiB flash, so not always last sector of flash
  - Layout
    - 2 bytes: magic value
    - 1 byte: start threshold
    - 1 byte: end threshold

Could increase it to reserve some space.

- Some kind of control byte?
- Space for second battery values?

EC should do bounds checking on the values when read even if the magic byte is valid and fall back to the defaults if it fails.

There's probably multiple things that set thresholds on boot now since they aren't persistent. EC should probably only write thresholds to flash if they've been changed to minimize the amount of writing to EC flash. It would mean having the distinction of "apply" (change is only in RAM) vs "commit" (change is written to flash).

crawfxrd, 2024-12-13:
Basic implementation done in #498.

closed 2024-12-13 by crawfxrd