system76/ec #137: Charging thresholds on lemp9

ozdreamern, 2021-01-07:
On my lemp9, I have no ability to control charging thresholds (`/sys/class/power_supply/BAT0/charge_control_{start,end}_threshold` don't exist).  I'm running stock Pop_OS 20.10, and all firmware and OS code seems up to date as far as I can tell.  The relevant EC code looks the same.  What am I missing?  Thanks for your time!

chronos000, 2021-01-14:
Same here

lemp9
Pop_OS 20.10
firmware 2020-09-17_f10af76

crawfxrd, 2021-01-14:
There has been no firmware release with charging thresholds for lemp9. You need 2020-10-02_2b077b5 or newer.

ozdreamern, 2021-01-15:
Ok, good to know it's not just my unit.  I hope there's an official release with support coming.  Or I wish I had time to learn the toolchain to build it myself.

ozdreamern, 2021-02-08:
Just want to say thank you for pushing this release out!

crawfxrd, 2021-03-23:
Available on lemp9 since 2021-01-19_d6d37c0.

closed 2021-03-23 by crawfxrd