system76/ec #144: darp7 - keystrokes "ca" cause a screenshot

mkellyxp, 2021-02-03:
As strange as it sounds, this brand new darp7 with a fresh pop 20.10 install with no extra software installed (except updates), will trigger a screenshot everytime i type "ca" in any program.. or while just at the desktop.  I don't have any keyboard shortcuts set up or anything like that.

I took a short video showing the issue

https://vimeo.com/508026110/188077da16

mkellyxp, 2021-02-03:
Can also confirm that this happens in a live USB boot of Manjaro KDE.. seeming to imply this is not a POP issue, but a hardware one

jackpot51, 2021-02-03:
@mkellyxp can you run the following commands to try and identify how this is happening? Let me know if there are errors:

```
git clone https://github.com/system76/ec
cd ec
./ectool.sh console
```

After the last command, produce the issue, then use CTRL+C to exit. The last few lines of the last command will help to identify the issue

mkellyxp, 2021-02-03:
Ok.. here we go!

mkelly@pop-os:~/ec$ ./ectool.sh console
    Finished release [optimized] target(s) in 0.00s
KB 15, 2, 0 = 0x005A, 0
BAT 8525 mV -2986 mA
KB 16, 2, 0 = 0x0021, 1
KB 12, 2, 0 = 0x001C, 1
BAT 8516 mV -2961 mA
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 0
KB 16, 2, 0 = 0x0021, 1
KB 12, 2, 0 = 0x001C, 1
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 0
BAT 8521 mV -2976 mA
KB 16, 2, 0 = 0x0021, 1
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 1
KB 16, 2, 0 = 0x0021, 1
KB 12, 2, 0 = 0x001C, 0
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 1
KB 13, 2, 0 = 0x2001, 1
KB 13, 2, 0 = 0x2001, 0
KB 16, 2, 0 = 0x0021, 1
KB 12, 2, 0 = 0x001C, 0
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 1
KB 16, 2, 0 = 0x0021, 1
KB 12, 2, 0 = 0x001C, 0
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 1
KB 16, 2, 0 = 0x0021, 1
KB 12, 2, 0 = 0x001C, 0
BAT 8500 mV -3656 mA
KB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 1
KB 12, 2, 0 = 0x001C, 0
BAT 8531 mV -3109 mA
KB 0, 6, 0 = 0x0014, 1
KB 16, 2, 0 = 0x0021, 1
KB 16, 2, 0 = 0x0021, 0
KB 0, 6, 0 = 0x0014, 0
BAT 8641 mV -1143 mA
KB 0, 6, 0 = 0x0014, 1
BAT 8642 mV -1100 mA
KB 0, 6, 0 = 0x0014, 0
BAT 8636 mV -1101 mA
KB 0, 6, 0 = 0x0014, 1
KB 16, 2, 0 = 0x0021, 1
^C

mkellyxp, 2021-02-03:
(just a little heads up too, i did have to install something called "cargo" to run this)  Easy fix.. but might trip up others

mkellyxp, 2021-02-03:
And oddly enough it seemed harder to make it do it this time.. but still did it.  very weird

ipaqmaster, 2021-02-03:
I also encountered the c+a screenshot problem on my darp7 last night, brand new install with a en_US keyboard selection.

I also noticed that the keys a+e+r causes a `0` to also be printed in the 4th position. This kind of stuff noticeable when touch-typing too quickly.

My darp7 isn't doing it this morning after initially thinking it may have been an unlabelled Fn+<something> combination causing it. So now I wonder if it may be heat related and only occurring a while after a 'cold' boot?

mkellyxp, 2021-02-03:
Strange. Mine actually seemed to get better after being more warmed up this morning. And was doing it last night and this morning. 

I took the back off mine and reset the keyboard ribbon cables. So I'll test more tonight.

ipaqmaster, 2021-02-03:
Ok I'll give the keyboard ribbon a re-seat and see if that helps me out too

mkellyxp, 2021-02-04:
Just fired up and it's doing it again consistently.. and here is the log

mkelly@pop-os:~/ec$ ./ectool.sh console
    Finished release [optimized] target(s) in 0.02s
[sudo] password for mkelly: 
KB 15, 2, 0 = 0x005A, 0
BAT 8771 mV 0 mA
BAT 8770 mV 0 mA
KB 16, 2, 0 = 0x0021, 1
cKB 16, 2, 0 = 0x0021, 0
KB 12, 2, 0 = 0x001C, 1
aBAT 8769 mV 0 mA
KB 12, 2, 0 = 0x001C, 0
KB 16, 2, 0 = 0x0021, 1
cKB 12, 2, 0 = 0x001C, 1
aBAT 8769 mV 0 mA
KB 13, 2, 0 = 0x2001, 1
KB 13, 2, 0 = 0x2001, 0
KB 12, 2, 0 = 0x001C, 0
KB 16, 2, 0 = 0x0021, 0
BAT 8771 mV 0 mA
KB 0, 6, 0 = 0x0014, 1
BAT 8770 mV 0 mA
KB 16, 2, 0 = 0x0021, 1
^C
mkelly@pop-os:~/ec$

mkellyxp, 2021-02-04:
and i can confirm that he a+e+r thing sometimes does 0 for me too  but way more rare than the ca thing

jackpot51, 2021-02-04:
Thanks for the logs, I'll be looking into it

mkellyxp, 2021-02-04:
Thanks @jackpot51 !  Hoping this can be just a firmware update or something.  Appreciate your time looking into it.  Let me know if you need me to test anything else.  This isn't a daily driver for me right now so I'll leave the software as-is

jacobgkau, 2021-02-04:
I've seen this happen on-and-off when typing `ch`; the numpad 3 is input. Re-seating the keyboard ribbon cable does temporarily fix the issue (I initially thought it was something wrong with the keyboard.)

If I type `P@` (that's capital P and an @ sign) at the same time, I see the regular 6 key input repeatedly, along with the spacebar slightly less often. (The 6 turns into a ^ because the shift key is held down.)

I haven't seen `aer` or `ac` cause any issues on the unit I have (but they could just be rarer on this particular unit.)

jackpot51, 2021-02-05:
I believe we have fixed this in https://github.com/system76/ec/commit/18ec0eb6948c3ea76ddfe2719a6b256033c21e4c. If anyone is brave enough to test beta firmware, please use Popsicle to flash the darp7-keyboard-fix.img file inside the [darp7-keyboard-fix.zip](https://github.com/system76/ec/files/5934097/darp7-keyboard-fix.zip) file to a USB stick. Then boot from the USB stick, this will walk you through the firmware update and power off the machine. When you turn it back on, after memory training for about 30 seconds, if you go to the firmware menu with ESC, it should list the firmware version as 2021-02-05_c3994a1. Now you can test the keyboard. Sequences that failed before for me were:

- Pressing c+a+r at the same time caused a phantom screenshot press
- Pressing shift+p+2 caused a phantom space press

mkellyxp, 2021-02-05:
@jackpot51 , I think you nailed it. After 10 minutes of trying to get it to trip up, it won't. Seems to work great!

Only note to others doing this fix. After you flash from firmware, when you reboot, pop_os will say you have a firmware update. 

If you're a fool like me and do it, it will roll you back to the old firmware. So I assume just ignore the firmware upgrade notification until this update is mainlined

ipaqmaster, 2021-02-06:
I also applied it a few hours ago and the keyboard issue hasn't come back yet either. Very nice

jackpot51, 2021-02-07:
Cool, we will release the update Monday to make sure support is there for other customers with questions

mkellyxp, 2021-02-08:
Good stuff @jackpot51 thanks again for the quick fix.  Def loving the darp7 alot..

closed 2021-02-08 by jackpot51