system76/ec #170: galp5: fans remain at 100% during suspend

leviport, 2021-03-24:
If I set the fans at 100% with the Fn + 1 hotkey and suspend the machine, it will suspend, but the fans stay spinning at 100% while it sleeps. I can also turn them on and off with the machine suspended without it waking up. I think the hotkey should probably be ignored while suspended, and the fans should probably turn off when entering suspend.

jackpot51, 2021-03-24:
Does this affect other models?

leviport, 2021-03-24:
Oryp7 doesn't do it. The machine wakes up when I hit Fn, so I can't turn them on while it's asleep. They also turn off when entering suspend, although resuming briefly makes them spin up fast for half a second before returning to the default mode. 

If it's related to SI0x, I'm betting darp7 and lemp10 would do the same thing.

jackpot51, 2021-03-24:
I think it is related to S0ix. Does the power LED pulse as though it is suspended while this is happening?

leviport, 2021-03-24:
Yes, though I see the sequence reset when I hit the hotkey.

jacobgkau, 2021-03-24:
> If it's related to SI0x, I'm betting darp7 and lemp10 would do the same thing.

Just to confirm, I can recreate this on darp7 and lemp10.

PR #172 (2021-04-02): Fixes for S0iX systems
- [x] Fix fan_max keeping fan on when in S0iX - fixes #170 
- [x] Report all keys as released when lid is closed - fixes #169
closed 2021-04-02 by jackpot51