system76/ec #174: Keyboard colors broken on oryp 6/7

MilesBHuff, 2021-04-12:
*Cross-post from https://github.com/pop-os/system76-acpi-dkms/issues/8.*

**Distribution (run `cat /etc/os-release`):**
```
NAME="Manjaro Linux"
ID=manjaro
ID_LIKE=arch
BUILD_ID=rolling
PRETTY_NAME="Manjaro Linux"
ANSI_COLOR="32;1;24;144;200"
HOME_URL="https://manjaro.org/"
DOCUMENTATION_URL="https://wiki.manjaro.org/"
SUPPORT_URL="https://manjaro.org/"
BUG_REPORT_URL="https://bugs.manjaro.org/"
LOGO=manjarolinux
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
`system76-acpi-dkms 1.0.2+1+g0bc966c-1`

**Issue/Bug Description:**
Setting the keyboard of the oryp7 to anything other than `ffffff` results in completely random colors being assigned.

**Steps to reproduce (if you know):**
Set keyboard brightness to anything less than 100% (I use 20%):
```
sudo su -c 'echo 667788 > /sys/class/leds/system76_acpi::kbd_backlight/color'
# Repeat the above ad nauseum
```

**Expected behavior:**
The setting exposed by `system76-acpi-dkms` should use the assigned hex color.

**Other Notes:**

MilesBHuff, 2021-04-24:
Any updates on this?  Keyboard brightness is completely broken, lol.  Defeats the purpose of having brightness control buttons on the numpad.

jacobgkau, 2021-04-26:
> Keyboard brightness is completely broken, lol.

That is an exaggeration; brightness works normally with all predefined colors. The issue is that some non-predefined colors are having the color changed along with the brightness, which is an issue that should be fixed, but is not as severe as not being able to change brightness at all.

Looking at this some more, the issue is greatly reduced when using a fully-valued color. As a workaround until this is fixed, I would recommend using a color picker like [this one](https://html-color-codes.info/old/colorpicker.html) to find the color you're wanting, making sure to leave the dot in the top right of the square (or at least along the top edge.) While I am still seeing a small variation happening with some colors (like yellow and purple shades), it seems adjusting brightness still keeps the overall color when I only use colors with a value of 100%.

Alternatively, if you have a specific brightness you're wanting to achieve, you can set the keyboard backlight to 100% and drag the dot down vertically to mix black into your color code. Reducing the value percentage in the color code will reduce brightness, even if the backlight is set to 100% power in the firmware. (This workaround is mutually exclusive to the previous one, since lowering the value is what triggers the drastic color changes on firmware brightness change.)

MilesBHuff, 2021-04-26:
> That is an exaggeration; brightness works normally with all predefined colors. The issue is that some non-predefined colors are having the color changed along with the brightness, which is an issue that should be fixed, but is not as severe as not being able to change brightness at all.

Ah!  I see.  Thank you for that clarification.  I'm using [a program I wrote](https://github.com/MilesBHuff/Ambient-Keyboard-Lighting) that automatically sets the keyboard to the average color of the display, and since it uses non-fully-valued colors almost exclusively, the keyboard seems completely broken when I turn the brightness up and down.  If it works for all predefined colors, then yeah, this is clearly not "completely broken", and is really probably not even a priority.  I'll just make do for the time-being.  Thanks for the clarification.  :)

MilesBHuff, 2022-10-11:
Maybe the issue has been fixed by chance since I filed the ticket;  I'll try to reproduce with the latest ec.

crawfxrd, 2022-10-11:
It's not fixed. Setting PWM to inhibit mode while changing the values also doesn't affect the behavior.

crawfxrd, 2022-10-11:
Might have a fix.

This should be reproducible on all board using IT5570E. What about boards using IT8587E (addw1, oryp5)?

crawfxrd, 2022-10-11:
@jacobgkau can you check what the behavior on oryp5 or addw1 is? The IT8587E doesn't have (or at least doesn't document) the register for the fix in #304.

jacobgkau, 2022-10-11:
I don't have either of those machines at the moment, but I'll coordinate with the team to check that.

crawfxrd, 2022-10-12:
Actually, only addw1 uses PWM (oryp5 uses a stupid I2C config), so it won't matter since we're not making a release for addw1 in this batch.

PR #304 (2022-10-14): pwm: Reload counters when they reach 0
Add IT5570E register that controls when the down counters are updated. Set them to update when they reach 0, instead of immediately when DCRi is written.

Fixes keyboard color changing when changing brightness levels if not using 0xFF for an RGB value.

Fixes: #174
closed 2022-10-14 by crawfxrd