system76/ec #181: Oryx Pro Fan/s not working

thejoker3000, 2021-04-23:
I got my Oryx Pro yesterday or day before and after running all the updates and firmware update that there was, I noticed today that the fan/s were not working. I'm not sure if it has something to do with the settings or not. I have it set to:
- Integrated graphics
- Balanced 

I tried searching for information on this but could only find issues on fan spikes. However, this is not my concern.
I'm new to system76 and loving it but learning how to use it. Not a Linux n00bie.

```
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os
```

thejoker3000, 2021-04-23:
Also, if they don't work with **integrated graphics**, the heating of the laptop doesn't bother me but can to some. So I was wondering also if it would be a good idea to have one at least working just to keep it a bit cooler.

leviport, 2021-04-23:
Do your fans turn on if you hit `Fn` + `1`? That's `1` on the number row next to `~', not on the number pad. Hit the same hotkey again to return the fans to normal. If you turn the keyboard backlight on, you can see whether each fan is spinning through the vents in the back panel.

The fans on the Oryx are tuned to stay pretty quiet unless it really needs the cooling. The all metal chassis means it can feel a bit toasty to the touch, but that doesn't necessarily mean it is overheating. As long as your fans kick up to 100% with that hotkey, this is probably to be expected. 

If both fans don't spin up to 100% when you toggle that hotkey, I suggest reaching out to [support](https://support.system76.com/), and they will help get your machine repaired.

thejoker3000, 2021-04-23:
I didn't know about the hotkey. Thanks for pointing that out. 
So to be clear, if I press that hotkey and leave it on, will the fans stay on at 100% or does it get quiet after time once the bottom metal chassis cools down?

leviport, 2021-04-23:
They stay at 100% until the hotkey is hit again. That hotkey is mostly for testing fans.

thejoker3000, 2021-04-23:
OK.
So do you have to have it set to another graphics option for them to be running constantly but quiet? I mean, because they work if I press the hotkey, but stay off when I don't press `FN`+`1`

MilesBHuff, 2021-04-24:
@thejoker3000 Try the fan curve in #179.

thejoker3000, 2021-04-24:
@MilesBHuff How do I do that? Editing the firmware or whatever this is I don't know how to do. Or is this an option in the system somewhere?

MilesBHuff, 2021-04-24:
@thejoker3000 The fact that you're asking how to do this is a sign that you probably shouldn't do this, since it involves merging some PRs into a local clone of this repo, and then compiling and flashing a custom EC.

thejoker3000, 2021-04-24:
Yeah... I'd prefer not touching that. I kind of though it was that, but just prefer not touching that until it comes out in an update. I will however just use the hotkey to enable the fans every so often to keep it cool.

MilesBHuff, 2021-04-24:
@thejoker3000 It's highly unlikely that #179 will ever be merged;  System76 likes the fans to come on as rarely as possible, per #9.  Your best hope is that https://github.com/pop-os/system76-acpi-dkms/issues/9 gets implemented.  Until then, flashing a custom EC is the only way to improve the fan curve.

thejoker3000, 2021-04-24:
OK, so looks like system76-acpi-dkms would be the best option when it gets implemented like you said. 
Do you know if I can install and test it out yet or is it not fully ready?

MilesBHuff, 2021-04-24:
@thejoker3000 `system76-acpi-dkms` is stable and only needs to be installed explicitly if you are not using Pop!_OS (On Pop!_OS, it comes baked into the kernel, from what I understand.).  However, the aforementioned feature is not yet under development, so the files exposed by `system76-acpi-dkms` in sysfs will be immutable.

thejoker3000, 2021-04-24:
OK I see. Thanks for the help @leviport and @MilesBHuff Appreciate it very much.
Going to close this. Reopen if needed.

closed 2021-04-24 by thejoker3000

thejoker3000, 2021-05-30:
I think I found another way that could help with this, https://github.com/vmatare/thinkfan
I've been using a laptop cooling pad which has helped keep the temp. low

thejoker3000, 2021-05-30:
OK, so thinkfan (from above) doesn't work. At least for me. So I found this: https://github.com/davidrohr/clevo-indicator.git
It says it's for Clevo laptops but I think if you modify the source code, it can be used for system76