system76/ec #182: Dependency/Discussion: remove usage of `sudo` in scripts/makefiles

eyJhb, 2021-04-23:
After reading some PRs, I can see more and more `sudo` commands getting introduced into the code.
This can be a issue, seeing as more and more people are switching away from `sudo` and using such things as `doas` instead.

I propose removing `sudo` from the code, and making it either configurable with a variable, or telling the user to please run X script as the root user.

If I find the time, I would like to take a stab at this :)

Any comments on this are appreciated!

crawfxrd, 2021-04-23:
> After reading some PRs, I can see more and more sudo commands getting introduced into the code.

Such as which ones? I don't see any current PRs adding sudo.

> more and more people are switching away from sudo and using such things as doas instead.

Do you have any sources for that?

sudo is a pretty standard tool available on every Linux distro, whereas doas is an OpenBSD tool.

> I propose removing `sudo` from the code

Installing dependencies, accessing certain sysfs values, and LPC/SPI access all require root access to work.

> or telling the user to please run X script as the root user.

That would require every script to be run as root instead of only the commands that need root access, and probably breaks `deps.sh`.

crawfxrd, 2021-04-26:
> you removed "and making it either configurable with a variable", which is pretty important.

I am simply dismissing the option of being able to remove the use of sudo entirely.

> Then you have left one option left I suppose, ie. `and making it either configurable with a variable`.

Yes. This is the only way it would make sense.

But, again, doas is an OpenBSD tool. (And I don't expect OpenBSD devs are the ones maintaining the Linux fork.) How many people use it? What value is there to maintain compatibility with it?

eyJhb, 2021-04-27:
> I am simply dismissing the option of being able to remove the use of sudo entirely.
I was never saying that as a option, but maybe the comma made it seem like that. Sorry :)

> But, again, doas is an OpenBSD tool. (And I don't expect OpenBSD devs are the ones maintaining the Linux fork.) How many people use it? What value is there to maintain compatibility with it?

Not sure how many have switched, since I guess it is not that easy on most distros.
I am running NixOS which makes it quite easy to do this.

However, this is only a suggestion and a nice-to-have thing, and shouldn't be a priority :)
Guessing it could be a nice "first-issue" for somebody to pick up. :)

ids1024, 2021-04-27:
I suppose in `Makefile`s it's easy enough to have `SUDO := sudo` so a different command can be passed as an argument. Not sure about shell scripts.