system76/ec #183: "make" failing: "No such file or directory"

larsblumberg, 2021-04-25:
Hi!

I am running `make` with `config.mk` having these settings:

```
BOARD?=system76/lemp10
KEYMAP?=lars
```

Also copied `src/board/system76/lemp10/keymap/` to `lars.c` within the same folder.

Unfortunately `make` is failing with:

```
sdcc -mmcs51 --model-small --code-loc 1024 --code-size 1024 --Werror -Isrc/common/include -D__FIRMWARE_VERSION__=2021-04-02_16778e4-dirty -Isrc/board/system76/lemp10/include -D__BOARD__=system76/lemp10 -DEC_ESPI=1 -DKBLED_DAC=2 -DI2C_SMBUS=I2C_4 -DPS2_TOUCHPAD=PS2_3 -DCHARGER_CHARGE_CURRENT=1536 -DCHARGER_CHARGE_VOLTAGE=8800 -DCHARGER_INPUT_CURRENT=1600 -DBATTERY_START_THRESHOLD=0 -DBATTERY_END_THRESHOLD=100 -DLEVEL=4 -Isrc/board/system76/common/include -DSCRATCH_OFFSET=1024 -DSCRATCH_SIZE=1024 -Ibuild/system76/lemp10/2021-04-02_16778e4-dirty/include -DFLASH_OFFSET=2048 -DFLASH_SIZE=1024 -Ibuild/system76/lemp10/2021-04-02_16778e4-dirty/include -Isrc/ec/it5570e/include -D__EC__=it5570e -Isrc/arch/8051/include -D__ARCH__=8051 -Isrc/board/system76/common/scratch/include -D__SCRATCH__ -o build/system76/lemp10/2021-04-02_16778e4-dirty/scratch/board/system76/common/scratch/main.rel -c src/board/system76/common/scratch/main.c
make: sdcc: No such file or directory
make: *** [src/board/system76/common/scratch/scratch.mk:50: build/system76/lemp10/2021-04-02_16778e4-dirty/scratch/board/system76/common/scratch/main.rel] Error 127
```

What am I missing?

crawfxrd, 2021-04-25:
You need to install the dependencies.

closed 2021-04-25 by crawfxrd

larsblumberg, 2021-04-25:
Thanks a lot for the quick reply!

I've created a pull request https://github.com/system76/ec/pull/184 so that other users don't run into the same issue. People who are just reading the keyboard customization instructions might not be aware of the readme file in the root directory which lists an additional build step, namely how to install the build dependencies.