system76/ec #185: Missing dependencies when running `make flash_internal` on a (fresh) PopOS 20.10

larsblumberg, 2021-04-25:
Hi!

I did run `scripts/deps.sh` and `make flash_internal`.

Firstly I got an error that `cargo` wasn't available (I'm going to create a Pull Request to add it to the build dependencies installation script) and then I got the following error:

```
$ RUST_BACKTRACE=1 make flash_internal
[...]
cargo build --manifest-path tool/Cargo.toml --release
   Compiling hidapi v1.2.5
error: failed to run custom build command for `hidapi v1.2.5`

Caused by:
  process didn't exit successfully: `/home/lars/coding/system76-ec-firmware/tool/target/release/build/hidapi-67d05b15b6f31fa6/build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'Unable to find libudev: Failure { command: "\"pkg-config\" \"--libs\" \"--cflags\" \"libudev\"", output: Output { status: ExitStatus(ExitStatus(256)), stdout: "", stderr: "Package libudev was not found in the pkg-config search path.\nPerhaps you should add the directory containing `libudev.pc\'\nto the PKG_CONFIG_PATH environment variable\nNo package \'libudev\' found\n" } }', /home/lars/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.2.5/build.rs:53:54
  stack backtrace:
     0: rust_begin_unwind
               at /build/rustc-71Ymwo/rustc-1.47.0+dfsg1+llvm/library/std/src/panicking.rs:475
     1: core::panicking::panic_fmt
               at /build/rustc-71Ymwo/rustc-1.47.0+dfsg1+llvm/library/core/src/panicking.rs:85
     2: core::result::unwrap_failed
               at /build/rustc-71Ymwo/rustc-1.47.0+dfsg1+llvm/library/core/src/result.rs:1220
     3: build_script_build::compile_linux::{{closure}}
     4: build_script_build::main
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
make: *** [src/board/system76/common/common.mk:51: flash_internal] Error 101
```

The env var `PKG_CONFIG_PATH` is empty on my machine (Latest PopOS on a lemur10).

The missing file `libudev.pc` is located here:

```
$ find / -type f -name "libudev.pc" 2>/dev/null
/usr/lib/x86_64-linux-gnu/pkgconfig/libudev.pc
```.

After running

```
RUST_BACKTRACE=1 PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig/ make flash_internal
```

(TBC)

crawfxrd, 2021-04-26:
This error happens when `libudev-dev` is not installed. `PKG_CONFIG_PATH` does not need to be modified.

larsblumberg, 2021-04-26:
Mhm, I'm using PopOS 20.10, it either ships with `libudev-dev` (or got installed via `deps.sh`), however `PKG_CONFIG_PATH` is unset/empty on my machine.

I'm happy with my above workaround as I can't say for sure what the root cause of the problem is.

larsblumberg, 2021-05-21:
I just encountered the same issue again. Did run `./scripts/deps.sh` again before building.

`PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig/ make flash_internal` fixes it.

crawfxrd, 2021-05-21:
`PKG_CONFIG_PATH` _should_ be empty. It should _not_ be modified.

Looking at your backtrace again, you are using the wrong version of `cargo` (distro-provided package).

If you are in the same windows from running `scripts/deps.sh`, did you source `~/.cargo/env`?

larsblumberg, 2021-05-23:
Here's the output of installing the dependencies, sourcing the env vars, calling make to flash and the output the two commands you have listed above.

`pgk-config` says that `libudev-dev` is not installed.
`apt list` however lists `libudev-dev` as installed.

I don't have enough knowledge about `pkg-config` to judge what this means.

```
$ ./scripts/deps.sh
Installing system build dependencies
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libudev-dev is already the newest version (246.6-1ubuntu1.3pop0~1617377719~20.10~ce920ac).
make is already the newest version (4.3-4ubuntu1).
avr-libc is already the newest version (1:2.0.0+Atmel3.6.1-2).
avrdude is already the newest version (6.3-20171130+svn1429-2).
gcc-avr is already the newest version (1:5.4.0+Atmel3.6.1-2build1).
sdcc is already the newest version (4.0.0+dfsg-2).
curl is already the newest version (7.68.0-1ubuntu4.3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Initializing submodules
Installing pinned Rust toolchain and components
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/lars/.rustup

nightly-2020-07-27-x86_64-unknown-linux-gnu (overridden by '/home/lars/coding/system76-ec-firmware/rust-toolchain')
rustc 1.47.0-nightly (6c8927b0c 2020-07-26)
Successfully installed dependencies

$ source  $HOME/.cargo/env

$ make flash_internal
cargo build --manifest-path tool/Cargo.toml --release
   Compiling hidapi v1.2.5
error: failed to run custom build command for `hidapi v1.2.5`

Caused by:
  process didn't exit successfully: `/home/lars/coding/system76-ec-firmware/tool/target/release/build/hidapi-96aa10245de5b046/build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'Unable to find libudev: Failure { command: "\"pkg-config\" \"--libs\" \"--cflags\" \"libudev\"", output: Output { status: ExitStatus(ExitStatus(256)), stdout: "", stderr: "Package libudev was not found in the pkg-config search path.\nPerhaps you should add the directory containing `libudev.pc\'\nto the PKG_CONFIG_PATH environment variable\nNo package \'libudev\' found\n" } }', /home/lars/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.2.5/build.rs:53:54
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make: *** [src/board/system76/common/common.mk:51: flash_internal] Error 101

$ pkg-config --list-all | grep libudev

$ pkg-config --print-errors libudev
Package libudev was not found in the pkg-config search path.
Perhaps you should add the directory containing `libudev.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libudev' found

$ pkg-config --print-errors libudev-dev
Package libudev-dev was not found in the pkg-config search path.
Perhaps you should add the directory containing `libudev-dev.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libudev-dev' found

$ apt list | grep libudev-dev
libudev-dev/groovy,now 246.6-1ubuntu1.3pop0~1617377719~20.10~ce920ac amd64 [installed]
libudev-dev/groovy 246.6-1ubuntu1.3pop0~1617377719~20.10~ce920ac i386
```

crawfxrd, 2021-05-24:
At this point, I can only assume your environment is broken. Are you somehow using `pkg-config:i386`?

larsblumberg, 2021-05-24:
I don't think so, as I have a quiet fresh system that I set up only very recently (beginning of this year) -- didn't install much, most of the software through *Pop!_Shop* and some other through `brew`.

larsblumberg, 2021-05-24:
I am going to close this issue again as I am not expecting to solve it for now. I have a workaround (as noted above) which is okay.

closed 2021-05-24 by larsblumberg