system76/ec #191: Problem with build instructions

davidskeck, 2021-05-09:
Hello,

I am having an issue with the build instructions in `keyboard-layout-customization`.

I have followed the instructions as best as I know how, but I have a few issues.

I have a `config.mk` file in the `ec/` directory, and I am calling `make`.!

Here's the result of calling `make`:
[image]

But, this version does not match my currently installed version:
[image]

Even though I am on the right commit from `firmware-open`:

[image]

It does match the commit in `ec`. Is this the correct way to build, even though the version info seems to mismatch?

Thank you!

davidskeck, 2021-05-09:
I just tried flashing the result of the make instruction and my new keymap worked. I'm guessing the version number doesn't have to match the report from the System Firmware tool. The firmware version reported by that tool also hasn't changed. I'll leave this open in case anyone wants to verify that the mismatched versions are OK. So far I don't have any issues.

crawfxrd, 2021-05-10:
Yes. firmware-manager reports the system firmware version. This is the information reported in the DMI table, which you can see with `sudo dmidecode -t bios`.

The EC version, specifically, can be determined with
```
./scripts/ectool.sh info
```

- If building/flashing EC firmware from firmware-open, the version of the EC firmware is [set to match the version of the system firmware](https://github.com/system76/firmware-open/blob/0905c82add536432a7a621992881dadc65e14177/scripts/build.sh#L99)
- If building/flashing the EC firmware from this repo, then it will have a different version than the system firmware

davidskeck, 2021-05-10:
Thank you, @crawfxrd. That clears this up for me!

closed 2021-05-10 by davidskeck