system76/ec #213: Is it possible to un-hardcode fan curves?

Raikiri, 2021-06-09:
Right now the only way of controlling the fan curves is by changing the hardcoded values in src/board/system76/oryp7/board.mk , recompiling, flashing, etc. It's not particularly user-friendly.

Is there a technical possibility of using a user-provided curve (from RAM) if it's available and use a hardcoded fan-curve as a fallback for all other cases?

I'm aware that there's an issue for pop-os https://github.com/pop-os/system76-acpi-dkms/issues/9 that requests practically that but for pop-os exclusively. I think this issue deserves to have a resolution for all platforms.

MilesBHuff, 2021-06-12:
@Raikiri
> I'm aware that there's an issue for pop-os 

It's not just for Pop!_OS.  A sysfs interface would expose fan controls for all distros.  OP even explicitly mentions that they are running Manjaro.

Raikiri, 2021-06-12:
@MilesBHuff 
"all distros" != "all os's"

MilesBHuff, 2021-06-12:
That is true.  But I believe System76 recommends the proprietary BIOS when switching to Windows.  If so, that's a non-starter.  And who would put OSX on a Sys76 computer?  That just leaves edge cases, like BSD.

Raikiri, 2021-06-12:
i don't remember any recommendation related to using proprietary firmware with windows

MilesBHuff, 2021-06-12:
It seems I may be mistaken.  I thought I'd read that in a System76 support article;  but I am unable to find anything of the sort at the moment.

internalfx, 2021-10-01:
I would love to have a fan curve config file!

iburzynski, 2024-01-25:
I'd also like to see this feature. The hardcoded fan curve settings for `oryp8` are awful, and cause my fan to constantly flicker on and off around the 55 degree threshold during normal use. I'm not able to modify the curve and flash the firmware because I'm using NixOS and can't build the firmware successfully with cargo.

MilesBHuff, 2024-02-07:
@iburzynski 

> The hardcoded fan curve settings for `oryp8` are awful, and cause my fan to constantly flicker on and off around the 55 degree threshold during normal use.

They're just as bad on `oryp7`.  I have *much* better fan settings here, but System76 refuses to adopt them:  https://github.com/system76/ec/pull/179

> I'm not able to modify the curve and flash the firmware because I'm using NixOS and can't build the firmware successfully with cargo.

You don't need to build the entirety of the firmware;  you only need to flash the ec, which is *much* easier.

leviport, 2024-02-07:
We're not refusing anything. If we were, we would have closed your PR. We just haven't had time to test it yet.

Raikiri, 2024-02-08:
@MilesBHuff 
> but System76 refuses to adopt anything like them: https://github.com/system76/ec/pull/179
It's been just 3 years, give them some time.

crawfxrd, 2024-05-13:
Fans should only run when system is in S0, so possible solution to move the fan table from EC flash to system flash:

- Rewrite EC fan logic to only implement the mechanism for accessing duty/RPM value
  - Probably have EC turn on fans with some non-zero value in case firmware hangs
- Rewrite coreboot+EC ACPI access for writing the target values
- Implement edk2 driver+lib that reads/writes fan data in system NVRAM space (SMMSTORE) and controls fans through ACPI methods
- Implement edk2 app (:vomiting_face:) or system app that allows modifying the NVRAM data
- Update system76-acpi for sysfs access

There probably needs to be an "auto/manual" fan mode" setting in this to allow switching between firmware controlled fan points and writing ACPI/sysfs values from a system application.