system76/ec #223: How to define keycode for "scroll lock"?

larsblumberg, 2021-07-13:
I use the "Scroll Lock" key as the "compose key" to type "special characters" on Pop_OS! (configurable via Settings > Keyboard > Type Special Characters).

The firmware currently doesn't define key codes for the scroll lock key as documented here:

https://github.com/system76/ec/blob/c7fafe01038d3adbc9893cd5ed2214f5424563bb/src/common/include/common/keymap.h#L170

Is there a chance to find and define the corresponding key codes myself or is there more engineering effort involved than just defining the codes?

closed 2021-07-20 by jackpot51

crawfxrd, 2021-07-20:
Scroll lock can now be defined with `K_SCROLL_LOCK` in the keymap.

larsblumberg, 2021-07-23:
Thanks a ton @crawfxrd!

larsblumberg, 2021-08-13:
@crawfxrd Just wanted to confirm that I can type special characters now (such as German umlauts) using the Scroll Lock key on my lemp10. Thanks again for your changes once more.

PR #225 (2021-07-20): Add scroll lock to default keymaps
I don't know anything that uses this, so I only checked that xev reported the key as scroll lock.

Resolves: #223