system76/ec #236: Does firmware utility notify new firmware updates after I flashed a custom firmware?

larsblumberg, 2021-08-13:
Question: As I flashed a custom firmware in order to remap some keys,

(A) do I still get notified by the System76 Firmware Updater about new firmware updates so that I can rebase my fork onto the latest version and re-flash it?

or (B) do I need to manually check whether there are new firmware releases (=certain commits) available? If yes, how do I obtain this information?

crawfxrd, 2021-08-13:
If you flashed only the EC firmware, you won't get notified.

If you flashed the BIOS firmware, you will get notifications if you used _any_ commit other than the one used for the latest release, or a dirty commit. (Ref: system76/firmware-open#49)

larsblumberg, 2021-08-13:
Thanks @crawfxrd, this answers my question. Since I am only flashing the EC firmware according to https://github.com/system76/ec/blob/master/doc/keyboard-layout-customization.md, I assume that I am flash thing the EC firmware only, hence I am not getting any firmware update notifications on Pop_OS!.

I could manually check if there are new firmware versions available every now and then. Therefore I've just discovered https://github.com/system76/firmware-open/blob/master/CHANGELOG.md which keeps a list of firmware release dates. Is there a way to identify the related commit that I have to rebase my fork of the `ec` repository to?

PS: I am happy to document these steps somewhere as I guess all keyboard layout hackers suffer from that issue.

jacobgkau, 2021-08-13:
> Is there a way to identify the related commit that I have to rebase my fork of the `ec` repository to?

If you navigate to a `firmware-open` commit from the date of the changelog entry (and click the `Browse Files` button), then you can see which `ec` commit is included as a submodule:

[image]

crawfxrd, 2021-08-13:
> I assume that I am flash thing the EC firmware only, hence I am not getting any firmware update notifications on Pop_OS!.

Correct.

> Therefore I've just discovered CHANGELOG.md which keeps a list of firmware release dates

This keeps a list of _all_ releases, but not _model specific_ releases. Some of those releases don't exist on certain models. For example, 2021-07-20 is only available on gaze16. (Not yet released for other models.)

firmware-manager has a history of the model specific releases, which will include the commit hash to use. (A `changelog.json` is included in the tarball system76-firmware)

larsblumberg, 2021-08-13:
@crawfxrd I found these commands to work to receive the content of the `changelog.json`:

```
sudo apt update
sudo apt install system76-firmware
sudo system76-firmware-cli schedule
sudo cat /boot/efi/system76-firmware-update/firmware/changelog.json
sudo system76-firmware-cli unschedule
```

According to this output, `2021-04-07_236914e` is the latest release of lemp10.

@jackpot51 I'm afraid that these git sub modules point to the most recent commit on the main branch, not to the commit of the most recent. I assume that System76 considers a firmware to be ready only at the officially released commits, not during the commits in between. I might be wrong here!

jacobgkau, 2021-08-13:
> I'm afraid that these git sub modules point to the most recent commit on the main branch, not to the commit of the most recent.

I'm not sure I follow you. I can see that the submodule commit referenced is changed when the submodule is updated. My understanding is that the `ec` commit referenced is the version of `ec` that was included at the time the official firmware release based on that `firmware-open` commit was built. So when `236914e` was built, `ec` would have been version 16778e4:

[image]

> I assume that System76 considers a firmware to be ready only at the officially released commits, not during the commits in between. I might be wrong here!

We try to make sure that the master branch is always able to be used, although more extensive testing often doesn't happen until official releases.

larsblumberg, 2021-08-16:
@jacobgkau That makes totally sense. Thanks for clarifying! Once I know the commit hash of the release in `firmware-open` I can then obtain the commit hashes of the submodules as you laid. That's a strategy I am familiar with.

closed 2021-08-16 by larsblumberg

larsblumberg, 2021-09-20:
Today I got a system notification that there was a new firmware update although I am running customization of an older firmware:

[image]

[image]

That's good news as I am able to rebase my customization on top of the latest release once its made available.