system76/ec #239: Build instructions seem wrong: no KEYBOARD?

jbrains, 2021-08-29:
The build instructions for customizing a keymap seem wrong. Maybe they're out of date?

Specifically, this instruction fails for me:

    grep KEYBOARD src/board/system76/<model>/board.mk

There is no `KEYBOARD` match in `**/board.mk` in my BIOS revision `236914e` corresponding to this project's revision `16778e4`.

What should this instruction be now?

jbrains, 2021-08-29:
This part is also out of date now:

    cp src/keyboard/system76/<keyboard>/keymap/default.c src/keyboard/system76/<keyboard>/keymap/<custom>.c

Notably, rename `keyboard` to `board`.

jbrains, 2021-08-29:
Whereas I could figure out these instructions, it would be nicer if they were more clear:

```
Build the firmware specifying the KEYMAP you have added.

make BOARD=system76/lemp9 KEYMAP=custom

Use a config file to simplify this. In the EC project's root directory, create a file named config.mk. Define the BOARD and KEYMAP variables.

BOARD?=system76/lemp9
KEYMAP?=custom
```

Here, it would be nice to clarify that `lemp9` is `<model>` and that `custom` is `<custom>`, defined above. Either that or emphasize that this is an _example_ for the model `lemp9` and the custom layout filename `custom`.

jbrains, 2021-08-29:
> Then building the firmware only requires calling make.

    # Make will read config.mk to determine which board and keymap to use.
    make

Where?! The root of `ec` or the root of the larger firmware project? I assume it's the latter, but I would prefer to make that clear.

crawfxrd, 2021-09-08:
The instructions are correct for the current code. `16778e4` has the correct instructions for the pre-keyboard split.

closed 2022-06-30 by crawfxrd