system76/ec #247: Can't flash EC for oryp6

FengXi7420, 2021-09-29:
sudo ecflash/target/release/examples/isp build/system76/oryp6/2021-09-20_d3ee037/ec.rom
thread 'main' panicked at 'failed to flash: Error { kind: Unknown, description: "No such file or directory" }', examples/isp.rs:830:26
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make: *** [src/board/system76/common/common.mk:71：flash_external] 错误 101

crawfxrd, 2021-09-29:
Since you are running `flash_external`, do you have a Mega 2560 plugged it and does it create a `/dev/ttyACM0` device?

FengXi7420, 2021-09-29:
> Since you are running `flash_external`, do you have a Mega 2560 plugged it and does it create a `/dev/ttyACM0` device?

I'VE plugged the mega2560 but I'm not sure it has create a /dev/ttyACM0 device or not

FengXi7420, 2021-09-29:
but I'm not sure it has create a /dev/ttyACM0 device or not

jackpot51, 2021-10-04:
It looks like you solved this (https://github.com/system76/firmware-open/issues/255) so I will close this issue

FengXi7420, 2021-10-05:
> Since you are running `flash_external`, do you have a Mega 2560 plugged it and does it create a `/dev/ttyACM0` device?

Sorry, but how to generate GPIO config, how to use ecsim

jackpot51, 2021-10-05:
If you don't have a genuine System76 product, we cannot provide help.