system76/ec #250: oryp6 won't boot after flashing the ec

FengXi7, 2021-10-01:
predator@predator-Predator-PH717-72:~/ec$ make BOARD=system76/oryp6 flash_external
cargo build --manifest-path ecflash/Cargo.toml --example isp --release
warning: associated function is never used: `erase_chip`
   --> examples/isp.rs:225:12
    |
225 |     pub fn erase_chip(&mut self) -> Result<()> {
    |            ^^^^^^^^^^
    |
    = note: `#[warn(dead_code)]` on by default

warning: field is never read: `address`
   --> examples/isp.rs:449:5
    |
449 |     address: Pio<u8>,
    |     ^^^^^^^^^^^^^^^^

warning: field is never read: `data`
   --> examples/isp.rs:450:5
    |
450 |     data: Pio<u8>,
    |     ^^^^^^^^^^^^^

warning: associated function is never used: `super_io_read`
   --> examples/isp.rs:468:8
    |
468 |     fn super_io_read(&mut self, reg: u8) -> u8 {
    |        ^^^^^^^^^^^^^

warning: associated function is never used: `super_io_write`
   --> examples/isp.rs:473:8
    |
473 |     fn super_io_write(&mut self, reg: u8, value: u8) {
    |        ^^^^^^^^^^^^^^

warning: associated function is never used: `d2_read`
   --> examples/isp.rs:478:8
    |
478 |     fn d2_read(&mut self, reg: u8) -> u8 {
    |        ^^^^^^^

warning: associated function is never used: `d2_write`
   --> examples/isp.rs:483:8
    |
483 |     fn d2_write(&mut self, reg: u8, value: u8) {
    |        ^^^^^^^^

warning: associated function is never used: `i2ec_read`
   --> examples/isp.rs:488:8
    |
488 |     fn i2ec_read(&mut self, addr: u16) -> u8 {
    |        ^^^^^^^^^

warning: associated function is never used: `i2ec_write`
   --> examples/isp.rs:494:8
    |
494 |     fn i2ec_write(&mut self, addr: u16, value: u8) {
    |        ^^^^^^^^^^

warning: 9 warnings emitted

    Finished release [optimized] target(s) in 0.00s
sudo ecflash/target/release/examples/isp build/system76/oryp6/2021-09-20_d3ee037-dirty/ec.rom
Buffer size: 128
ID: 5570 VER: 2
SPI read
Saving ROM to backup.rom
SPI sector erase 000000
SPI sector erase 000400
SPI sector erase 000800
SPI sector erase 000C00
SPI sector erase 001000
SPI sector erase 001400
SPI sector erase 001800
SPI sector erase 001C00
SPI sector erase 002000
SPI sector erase 002400
SPI sector erase 002800
SPI sector erase 002C00
SPI sector erase 003000
SPI sector erase 003400
SPI sector erase 003800
SPI sector erase 003C00
SPI sector erase 004000
SPI sector erase 004400
SPI sector erase 004800
SPI sector erase 004C00
SPI sector erase 005000
SPI sector erase 005400
SPI sector erase 005800
SPI sector erase 005C00
SPI sector erase 006000
SPI sector erase 006400
SPI sector erase 006800
SPI sector erase 006C00
SPI sector erase 007000
SPI sector erase 007400
SPI sector already erased 007800
SPI sector already erased 007C00
SPI sector erase 008000
SPI sector erase 008400
SPI sector erase 008800
SPI sector erase 008C00
SPI sector erase 009000
SPI sector erase 009400
SPI sector erase 009800
SPI sector erase 009C00
SPI sector erase 00A000
SPI sector erase 00A400
SPI sector erase 00A800
SPI sector erase 00AC00
SPI sector erase 00B000
SPI sector erase 00B400
SPI sector erase 00B800
SPI sector erase 00BC00
SPI sector erase 00C000
SPI sector erase 00C400
SPI sector erase 00C800
SPI sector erase 00CC00
SPI sector erase 00D000
SPI sector erase 00D400
SPI sector erase 00D800
SPI sector erase 00DC00
SPI sector erase 00E000
SPI sector erase 00E400
SPI sector erase 00E800
SPI sector erase 00EC00
SPI sector erase 00F000
SPI sector erase 00F400
SPI sector erase 00F800
SPI sector erase 00FC00
SPI sector erase 010000
SPI sector erase 010400
SPI sector erase 010800
SPI sector erase 010C00
SPI sector erase 011000
SPI sector erase 011400
SPI sector erase 011800
SPI sector erase 011C00
SPI sector erase 012000
SPI sector erase 012400
SPI sector erase 012800
SPI sector erase 012C00
SPI sector erase 013000
SPI sector erase 013400
SPI sector erase 013800
SPI sector erase 013C00
SPI sector erase 014000
SPI sector erase 014400
SPI sector erase 014800
SPI sector erase 014C00
SPI sector erase 015000
SPI sector erase 015400
SPI sector already erased 015800
SPI sector already erased 015C00
SPI sector already erased 016000
SPI sector already erased 016400
SPI sector already erased 016800
SPI sector already erased 016C00
SPI sector already erased 017000
SPI sector already erased 017400
SPI sector already erased 017800
SPI sector already erased 017C00
SPI sector erase 018000
SPI sector erase 018400
SPI sector erase 018800
SPI sector erase 018C00
SPI sector erase 019000
SPI sector erase 019400
SPI sector erase 019800
SPI sector erase 019C00
SPI sector erase 01A000
SPI sector erase 01A400
SPI sector erase 01A800
SPI sector erase 01AC00
SPI sector erase 01B000
SPI sector erase 01B400
SPI sector erase 01B800
SPI sector erase 01BC00
SPI sector erase 01C000
SPI sector erase 01C400
SPI sector erase 01C800
SPI sector erase 01CC00
SPI sector already erased 01D000
SPI sector already erased 01D400
SPI sector already erased 01D800
SPI sector already erased 01DC00
SPI sector already erased 01E000
SPI sector already erased 01E400
SPI sector already erased 01E800
SPI sector already erased 01EC00
SPI sector already erased 01F000
SPI sector already erased 01F400
SPI sector already erased 01F800
SPI sector already erased 01FC00
SPI read
SPI AAI word program (accelerated)
  program 33932 / 33932
SPI read
Successfully programmed SPI ROM

After this, I connected my battery and AC cable, but there was no response after I press the power button. The charging LED was green and the power LED was orange

FengXi7, 2021-10-01:
Here's some debug info:
predator@predator-Predator-PH717-72:~/ec$ make BOARD=system76/oryp6 console_external
sudo test -c "/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85035303038351308102-if00"
sleep 1 && echo C | sudo tee "/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85035303038351308102-if00" &
sudo tio -b 1000000 -m INLCRNL -t "/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85035303038351308102-if00"
[tio 15:21:30] tio v1.32
[tio 15:21:30] Press ctrl-t q to quit
[tio 15:21:30] Connected
C
 [15:21:31] Entering console mode
[15:21:31] 
[15:21:37] System76 EC board 'system76/oryp6', version '2021-09-20_d3ee037-dirty'
[15:21:37] 00: power_on_ds5
[15:21:37] POWER_STATE_DS5
[15:21:37] Power adapter plugged in
[15:21:37] Charger enabled
[15:21:37] Battery:
[15:21:37]   Temperature: ERROR 0025
[15:21:37]   Voltage: ERROR 0025
[15:21:37]   Current: ERROR 0025
[15:21:37]   Charge: ERROR 0025
[15:21:37]   Status: ERROR 0025
[15:21:37] Charger (bq24780s):
[15:21:37]   ChargeOption0: 8108
[15:21:37]   ChargeOption1: C210
[15:21:37]   ChargeOption2: 0384
[15:21:37]   ChargeOption3: 1A40
[15:21:37]   ChargeCurrent: 0600
[15:21:37]   ChargeVoltage: 32F0
[15:21:37]   DishargeCurrent: 1800
[15:21:37]   InputCurrent: 1000
[15:21:37]   ProchotOption0: 4A54
[15:21:37]   ProchotOption1: 8120
[15:21:37]   ProchotStatus: 0020
[15:21:37] Lid open
[15:21:52] 81: Power switch press
[15:21:52] 81: power_on_s5
[15:21:52] VA_EC_EN = true
[15:21:52] DD_ON = true
[15:21:52] EC_RSMRST_N = true
[15:21:52] EC_EN = true
[15:21:52] POWER_STATE_S5
[15:21:57] 81: power_off_s5
[15:21:57] PM_PWROK = false
[15:21:57] EC_EN = false
[15:21:57] EC_RSMRST_N = false
[15:21:57] DD_ON = false
[15:21:57] VA_EC_EN = false
[15:21:57] POWER_STATE_DS5
[15:21:57] 84: Power switch release
[15:22:32] 30: Power switch press
[15:22:32] 30: power_on_s5
[15:22:32] VA_EC_EN = true
[15:22:32] DD_ON = true
[15:22:32] EC_RSMRST_N = true
[15:22:32] EC_EN = true
[15:22:32] POWER_STATE_S5
[15:22:37] 30: power_off_s5
[15:22:37] PM_PWROK = false
[15:22:37] EC_EN = false
[15:22:37] EC_RSMRST_N = false
[15:22:37] DD_ON = false
[15:22:37] VA_EC_EN = false
[15:22:37] POWER_STATE_DS5
[15:22:37] 33: Power switch release
[15:22:49] 4B: Power switch press
[15:22:49] 4B: power_on_s5
[15:22:49] VA_EC_EN = true
[15:22:49] DD_ON = true
[15:22:50] EC_RSMRST_N = true
[15:22:50] EC_EN = true
[15:22:50] POWER_STATE_S5
[15:22:55] 4B: power_off_s5
[15:22:55] PM_PWROK = false
[15:22:55] EC_EN = false
[15:22:55] EC_RSMRST_N = false
[15:22:55] DD_ON = false
[15:22:55] VA_EC_EN = false
[15:22:55] POWER_STATE_DS5
[15:22:55] 4E: Power switch release
[15:22:59] 42: Power switch press
[15:22:59] 42: power_on_s5
[15:22:59] VA_EC_EN = true
[15:22:59] DD_ON = true
[15:22:59] EC_RSMRST_N = true
[15:23:00] EC_EN = true
[15:23:00] POWER_STATE_S5
[15:23:05] 42: power_off_s5
[15:23:05] PM_PWROK = false
[15:23:05] EC_EN = false
[15:23:05] EC_RSMRST_N = false
[15:23:05] DD_ON = false
[15:23:05] VA_EC_EN = false
[15:23:05] POWER_STATE_DS5
[15:23:09] A8: Power switch release
[15:23:11] A2: Power switch press
[15:23:11] A2: power_on_s5
[15:23:11] VA_EC_EN = true
[15:23:11] DD_ON = true
[15:23:11] EC_RSMRST_N = true
[15:23:12] EC_EN = true
[15:23:12] POWER_STATE_S5
[15:23:17] A2: power_off_s5
[15:23:17] PM_PWROK = false
[15:23:17] EC_EN = false
[15:23:17] EC_RSMRST_N = false
[15:23:17] DD_ON = false
[15:23:17] VA_EC_EN = false
[15:23:17] POWER_STATE_DS5
[15:23:17] A5: Power switch release

FengXi7420, 2021-10-03:
Can't flash the bios using ch341a at this situation

crawfxrd, 2021-10-04:
You duplicated this issue at system76/firmware-open#255.

closed 2021-10-04 by crawfxrd