system76/ec #251: The oryp6 power lamp is not on

FTKyaoyuan, 2021-10-05:
We flashed the bios and ec of oryp6, but the power indicator did not light up when we started. We compared the pin function of gpio.c and our circuit diagram to be exactly the same, and we could not confirm the problem.

jacobgkau, 2021-10-05:
What's the System76 serial number of your machine? If you prefer not to provide it publicly, you are also welcome to open a [support ticket](https://support.system76.com/) instead if the power LED isn't working. Does it work if you revert back to the stable firmware?

jackpot51, 2021-10-05:
@FTKyaoyuan if you do not have a genuine System76 product, you should not create issues claiming there are problems with our firmware.

closed 2021-10-05 by jackpot51