system76/ec #256: [Draft] RFC: Persistent run-time configs

crawfxrd, 2021-11-13:
- Feature name: `persistent-configs`
- Start date: 2021-11-12
- Status: Draft

# Summary

Implement a data format and API for persistent run-time configuration data.

# Motivation

Allow users to set certain values at run-time so they do not have to recompile
and flash the EC firmware when they want to change them. Such values include:

- Battery charging thresholds (currently configurable, but not persistent)
- Webcam power
- Wireless power
- Fn lock
- Fan points

# Guide-level explanation

The EC can be configured to some extent during run-time using ectool, without
the need to recompile and flash new firmware.

A possible interface for ectool could be something like:

```
$ ectool config -h
Query or set EC run-time configuration data.

USAGE:
    ectool config [SUBCOMMAND]

OPTIONS:
    -h, --help      Prints help information

SUBCOMMANDS
    get             Get the data for a config
    set             Set the data for a config
    reset           Reset run-time configs and erase config flash space
```

```
$ ectool config get
01: charging-thresholds
    Values at while the battery will start and stop changing
    Start: 40
    End: 70

02: webcam-power
    Set webcam power on boot
    Power: OFF

03: wireless-power
    Set wireless power on boot
    Power: ON

04: fn-lock
    Use alternative functions when pressing F1-F12 by default
    FnLock: OFF

05: fan-points
    Fan curve
    Temp  Duty
    55    25
    65    30
    70    40
    75    60
    80    75
    85    90
[... 221 lines cut ...]

### Configure from system flash

As stated in #63, the UEFI payload could be responsible for configuring the EC
and storing values.

## Prior art

- [coreboot tables](https://github.com/coreboot/coreboot/blob/d1598991a3077f41c390151a1dd5832a003e5c1d/payloads/libpayload/include/coreboot_tables.h)
- [Multiboot2](https://www.gnu.org/software/grub/manual/multiboot2/multiboot.html)

# Unresolved questions

- Should the keyboard layout have a tag and be treated (internal to the config
  module) as a special config? Or should it remain separate with its own API?
- Should a 1 byte checksum field be added for each config to ensure that the
  data is valid when read on init?
- Should a 1 byte version field be added for each config to handle changes in
  semantics for a config?

# Future possibilities

- Fault tolerance mechanism (such as page flipping) so that if the config block
  becomes invalid it can reset to the previous init's config instead of erasing
  the block and resetting all values to defaults.

ids1024, 2022-01-05:
Hm, don't recall seeing this issue when it was created.

If flash space is very limited, and each tag has a known fixed address, using 2 bytes to store the signature and size seems potentially unnecessary. `config_get` could just take an offset in the config space and length, and `ectool` is responsible for knowing what is stored at that offset, and knowing that the size is correct.

Though that doesn't allow distinguishing per-tag whether or not it is set to anything.

The size of the config region could also be more flexible if it grew downward like a stack (when new tags are added), but I guess a single 1024 byte page is probably simpler to deal with.

> Should the keyboard layout have a tag and be treated (internal to the config module) as a special config? Or should it remain separate with its own API?

Not sure how that fits here, but it would be beneficial to be able to read or write the whole keymap in one operation.

MilesBHuff, 2022-05-23:
Just want to add that the goal of user-configurable fan curves can be achieved via a different method, which may be preferable: https://github.com/pop-os/system76-acpi-dkms/issues/9.
Advantages of this are that the fan curve can be changed without a reboot and there's no need to mess with limited NVRAM.
Downside is that the fan curve would only take effect once the system has booted.

crawfxrd, 2022-10-03:
Also may want to change the terminology to match coreboot. (Thinking about this since another idea is importing and using coreboot's Kbuild/Kconfig fork.)

- config: Value that is set at build-time
- option: Value that is _configurable_ at run-time

crawfxrd, 2022-10-13:
> Should the keyboard layout have a tag and be treated (internal to the config module) as a special config? Or should it remain separate with its own API?

Given the size of the data, and that there are 2 (custom and default), I think it would be better to keep in a separate region with its own header/format.

---

> Should a 1 byte checksum field be added for each config to ensure that the data is valid when read on init?

I think with the constraints of the 8051, this should not be done. If the header is valid, it should be assumed all entries are valid.

---

> Should a 1 byte version field be added for each config to handle changes in semantics for a config?

This should also not be done. If the semantics of a config changes significantly enough, it should use a new tag. If enough change, or the tags are exhausted, we can increment the version in the header and prune all deprecated tags.

---

Newly added:

> This was originally for ITE ECs. Should the ENE KB9548G used on HP Dev One be considered as well (4 KiB pages)?

I think the main difference (for this RFC) between the ITE and ENE ECs is the page size (1 KiB vs 4 KiB). This would affect what we use for the config block. Should it be 1024 bytes, or should it be 1 page (erase size)?

crawfxrd, 2023-02-27:
> ### Configure from system flash
> 
> As stated in https://github.com/system76/ec/pull/63, the UEFI payload could be responsible for configuring the EC and storing values.
>
> This may not work, as the EC must be able to read some values from the store before the SBIOS starts.

I wonder if it's possible to have the EC read configs from the system SPI flash. Configs could be stored in a RAW section of an EFI binary.

ilikenwf, 2024-03-06:
Whether it's defaulting webcam to off or just remembering the last set value, I'd love to see even just that change. The WIP pull request for backlight color/brightness works somewhat - it does remember if I have them on or off, however it usually resets the color to white if I remember correctly.

crawfxrd, 2024-04-27:
If it's possible to communicate with CMOS from the EC, that would be the best bet. RFC will have to be redone for it. Basically: Use coreboot CMOS option table, and offload as much as possible to system firmware.

crawfxrd, 2024-07-25:
Closing this in favor of pursuing https://github.com/system76/firmware-open/issues/571.

May be revisited once EC design improves for supporting configs that must be persistent without dependence on system firmware, such as battery charging thresholds

closed 2024-07-25 by crawfxrd

digitalcircuit, 2024-07-25:
Should we have a separate issue open for persistent battery charging thresholds?

I understand it not being a priority, just to keep track of it.  At the moment, I guide family members to "never shut down the laptop, just put it in suspend" to not lose the charging thresholds.

crawfxrd, 2024-07-25:
There is https://github.com/system76/ec/issues/128 for charging thresholds.