system76/ec #257: ectool forces static hidraw for hidapi crate

Conan-Kudo, 2021-11-14:
While [reviewing this package for inclusion in Fedora](https://bugzilla.redhat.com/show_bug.cgi?id=2022178), I noticed that it forces the `linux-static-hidraw` feature instead of the `linux-shared-hidraw` feature from the `hidapi` crate. In Fedora, we'd prefer that C libraries be dynamically linked for these kinds of things.

Please consider switching the requested feature of the `hidapi` crate to `linux-shared-hidraw`.

jackpot51, 2021-11-17:
@ids1024 was there a reason we used the `linux-static-hidraw` feature?

jackpot51, 2021-11-17:
https://github.com/system76/ec/commit/f4458aebcafefb3a9aea8bd18d6ede29cb192a04 was when this was added

ids1024, 2021-11-17:
I commented on https://github.com/pop-os/keyboard-configurator/issues/107 earlier.

The `hidapi` crate uses `linux-static-libusb` by default, so when I switched to the hidraw backend I left it with static linking. But dynamic linking should be fine.

jackpot51, 2021-11-18:
Released as system76_ectool version 0.3.7

PR #258 (2021-11-18): ectool: use linux-shared-hidraw feature for hidapi
Fixes #257
closed 2021-11-18 by jackpot51