system76/ec #260: Will there be any problems if the BIOS and ec flashing sequence is reversed?

FTKyaoyuan, 2021-12-11:
I first flashed in ec and then flashed in bios, but it can’t be turned on, the power light is flashing and the ec chip cannot be recognized. It may be that I have made a mistake in operation. Is there a remedy?

crawfxrd, 2021-12-11:
Assuming no components are damaged, it doesn't matter. But BIOS firmware and EC firmware must both be open firmware or both be proprietary firmware.

- Open EC firmware with proprietary BIOS firmware will never boot. We don't implement EC functionality that the proprietary BIOS requires.
- Open BIOS firmware with proprietary EC firmware only boots on some models.

But again, this isn't an official System76 laptop. So we can't really help with any problems you have with the unit.

>  Is there a remedy?

Flash back to proprietary firmware.

Did you ever attempt to [generate the GPIO config and power sequence](https://github.com/system76/ec/blob/master/doc/adding-a-new-board.md#gpios) from the proprietary firmware using [ecsim](https://github.com/system76/ecsim)?

FTKyaoyuan, 2021-12-11:
Thank you. I'll try

FTKyaoyuan, 2021-12-11:
It succeeded, thank you very much

closed 2021-12-11 by crawfxrd