system76/ec #269: Installing dependencies | "fatal: not a git repository (or any of the parent directories): .git"

TricksterSam, 2022-01-25:
When I run the dependencies installation script, I get an error when it attempts to initialize submodules. Any help?

```
▶ ./scripts/deps.sh
Installing system build dependencies
[sudo] password for samuelburt: 
Sorry, try again.
[sudo] password for samuelburt: 
Hit:1 http://us.archive.ubuntu.com/ubuntu impish InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu impish-security InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu impish-updates InRelease
Hit:4 http://us.archive.ubuntu.com/ubuntu impish-backports InRelease
Hit:5 http://apt.pop-os.org/proprietary impish InRelease
Hit:6 http://apt.pop-os.org/release impish InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libc6-dev' instead of 'libc-dev'
curl is already the newest version (7.74.0-1.3ubuntu2).
curl set to manually installed.
gcc is already the newest version (4:11.2.0-1ubuntu1).
libc6-dev is already the newest version (2.34-0ubuntu3).
libc6-dev set to manually installed.
make is already the newest version (4.3-4ubuntu1).
make set to manually installed.
xxd is already the newest version (2:8.2.2434-3ubuntu3.2).
The following packages were automatically installed and are no longer required:
  dzen2 libconfuse-common libconfuse2 libgarcon-1-0 libgarcon-common
  libxfce4ui-2-0 libxfce4ui-common libxfce4util-bin libxfce4util-common
  libxfce4util7 libxfconf-0-3 linux-headers-5.15.5-76051505
  linux-headers-5.15.5-76051505-generic linux-image-5.15.5-76051505-generic
  linux-modules-5.15.5-76051505-generic node-html5shiv xfconf
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  binutils-avr libftdi1 libhidapi-hidraw0 libhidapi-libusb0 libpkgconf3
  libusb-0.1-4 sdcc-libraries
Suggested packages:
  avrdude-doc gcc-doc python sdcc-ucsim
Recommended packages:
  sdcc-doc gputils
The following packages will be REMOVED:
  pkg-config
The following NEW packages will be installed:
  avr-libc avrdude binutils-avr gcc-avr libftdi1 libhidapi-dev
  libhidapi-hidraw0 libhidapi-libusb0 libpkgconf3 libudev-dev libusb-0.1-4
  pkgconf sdcc sdcc-libraries
0 upgraded, 14 newly installed, 1 to remove and 0 not upgraded.
Need to get 25.2 MB of archives.
After this operation, 158 MB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 libpkgconf3 amd64 1.7.4~git20210206+dcf529b-3 [27.9 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 pkgconf amd64 1.7.4~git20210206+dcf529b-3 [33.4 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu impish/main amd64 libusb-0.1-4 amd64 2:0.1.12-32build2 [17.8 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 libftdi1 amd64 0.20-4build8 [14.8 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 sdcc-libraries all 4.0.0+dfsg-2 [1,111 kB]
Get:6 http://apt.pop-os.org/release impish/main amd64 libudev-dev amd64 248.3-1ubuntu8.2pop0~1642619863~21.10~47b8972 [170 kB]
Get:7 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 sdcc amd64 4.0.0+dfsg-2 [1,690 kB]
Get:8 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 binutils-avr amd64 2.26.20160125+Atmel3.6.2-2 [1,492 kB]
Get:9 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 gcc-avr amd64 1:5.4.0+Atmel3.6.2-1 [15.4 MB]
Get:10 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 avr-libc all 1:2.0.0+Atmel3.6.2-1.1 [4,852 kB]
Get:11 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 libhidapi-libusb0 amd64 0.10.1+dfsg-1 [15.0 kB]
Get:12 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 avrdude amd64 6.3-20171130+svn1429-2 [316 kB]
Get:13 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 libhidapi-hidraw0 amd64 0.10.1+dfsg-1 [12.4 kB]
Get:14 http://us.archive.ubuntu.com/ubuntu impish/universe amd64 libhidapi-dev amd64 0.10.1+dfsg-1 [25.0 kB]
Fetched 25.2 MB in 1s (20.5 MB/s)              
Selecting previously unselected package libpkgconf3:amd64.
(Reading database ... 293641 files and directories currently installed.)
Preparing to unpack .../libpkgconf3_1.7.4~git20210206+dcf529b-3_amd64.deb ...
Unpacking libpkgconf3:amd64 (1.7.4~git20210206+dcf529b-3) ...
dpkg: pkg-config: dependency problems, but removing anyway as you requested:
 nvidia-settings depends on pkg-config.
 libostree-dev:amd64 depends on pkg-config.
 libjack-jackd2-dev:amd64 depends on pkg-config.
 libglib2.0-dev:amd64 depends on pkg-config.
 libflatpak-dev:amd64 depends on pkg-config.

(Reading database ... 293646 files and directories currently installed.)
Removing pkg-config (0.29.2-1ubuntu1) ...
Selecting previously unselected package pkgconf.
(Reading database ... 293632 files and directories currently installed.)
Preparing to unpack .../00-pkgconf_1.7.4~git20210206+dcf529b-3_amd64.deb ...
Adding 'diversion of /usr/bin/pkg-config to /usr/bin/pkg-config.real by pkgconf'
Adding 'diversion of /usr/share/aclocal/pkg.m4 to /usr/share/aclocal/pkg.real.m4 by pkgconf'
Adding 'diversion of /usr/share/man/man1/pkg-config.1.gz to /usr/share/man/man1/pkg-config.real.1.gz by pkgconf'
Adding 'diversion of /usr/share/pkg-config-crosswrapper to /usr/share/pkg-config-crosswrapper.real by pkgconf'
Unpacking pkgconf (1.7.4~git20210206+dcf529b-3) ...
Selecting previously unselected package libusb-0.1-4:amd64.
Preparing to unpack .../01-libusb-0.1-4_2%3a0.1.12-32build2_amd64.deb ...
Unpacking libusb-0.1-4:amd64 (2:0.1.12-32build2) ...
Selecting previously unselected package libftdi1:amd64.
Preparing to unpack .../02-libftdi1_0.20-4build8_amd64.deb ...
Unpacking libftdi1:amd64 (0.20-4build8) ...
Selecting previously unselected package libudev-dev:amd64.
Preparing to unpack .../03-libudev-dev_248.3-1ubuntu8.2pop0~1642619863~21.10~47b8972_amd64.deb ...
Unpacking libudev-dev:amd64 (248.3-1ubuntu8.2pop0~1642619863~21.10~47b8972) ...
Selecting previously unselected package sdcc-libraries.
Preparing to unpack .../04-sdcc-libraries_4.0.0+dfsg-2_all.deb ...
Unpacking sdcc-libraries (4.0.0+dfsg-2) ...
Selecting previously unselected package sdcc.
Preparing to unpack .../05-sdcc_4.0.0+dfsg-2_amd64.deb ...
Unpacking sdcc (4.0.0+dfsg-2) ...
Selecting previously unselected package binutils-avr.
Preparing to unpack .../06-binutils-avr_2.26.20160125+Atmel3.6.2-2_amd64.deb ...
Unpacking binutils-avr (2.26.20160125+Atmel3.6.2-2) ...
Selecting previously unselected package gcc-avr.
Preparing to unpack .../07-gcc-avr_1%3a5.4.0+Atmel3.6.2-1_amd64.deb ...
Unpacking gcc-avr (1:5.4.0+Atmel3.6.2-1) ...
Selecting previously unselected package avr-libc.
Preparing to unpack .../08-avr-libc_1%3a2.0.0+Atmel3.6.2-1.1_all.deb ...
Unpacking avr-libc (1:2.0.0+Atmel3.6.2-1.1) ...
Selecting previously unselected package libhidapi-libusb0:amd64.
Preparing to unpack .../09-libhidapi-libusb0_0.10.1+dfsg-1_amd64.deb ...
Unpacking libhidapi-libusb0:amd64 (0.10.1+dfsg-1) ...
Selecting previously unselected package avrdude.
Preparing to unpack .../10-avrdude_6.3-20171130+svn1429-2_amd64.deb ...
Unpacking avrdude (6.3-20171130+svn1429-2) ...
Selecting previously unselected package libhidapi-hidraw0:amd64.
Preparing to unpack .../11-libhidapi-hidraw0_0.10.1+dfsg-1_amd64.deb ...
Unpacking libhidapi-hidraw0:amd64 (0.10.1+dfsg-1) ...
Selecting previously unselected package libhidapi-dev:amd64.
Preparing to unpack .../12-libhidapi-dev_0.10.1+dfsg-1_amd64.deb ...
Unpacking libhidapi-dev:amd64 (0.10.1+dfsg-1) ...
Setting up binutils-avr (2.26.20160125+Atmel3.6.2-2) ...
Setting up libhidapi-hidraw0:amd64 (0.10.1+dfsg-1) ...
Setting up libusb-0.1-4:amd64 (2:0.1.12-32build2) ...
Setting up gcc-avr (1:5.4.0+Atmel3.6.2-1) ...
Setting up libpkgconf3:amd64 (1.7.4~git20210206+dcf529b-3) ...
Setting up libhidapi-libusb0:amd64 (0.10.1+dfsg-1) ...
Setting up libudev-dev:amd64 (248.3-1ubuntu8.2pop0~1642619863~21.10~47b8972) ...
Setting up sdcc-libraries (4.0.0+dfsg-2) ...
Setting up libhidapi-dev:amd64 (0.10.1+dfsg-1) ...
Setting up libftdi1:amd64 (0.20-4build8) ...
Setting up avr-libc (1:2.0.0+Atmel3.6.2-1.1) ...
Setting up sdcc (4.0.0+dfsg-2) ...
Setting up pkgconf (1.7.4~git20210206+dcf529b-3) ...
Setting up avrdude (6.3-20171130+svn1429-2) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.34-0ubuntu3) ...
Initializing submodules
fatal: not a git repository (or any of the parent directories): .git
Failed to install dependencies!
```

crawfxrd, 2022-01-25:
What does `git status` say?

TricksterSam, 2022-01-25:
Oh, same thing. I'll try cloning the repository rather than downloading and
opening the zip file. Still kind of new to Git.

Thanks, Tim.

TricksterSam, 2022-01-25:
That was definitely my problem. I didn't clone it, initially. Thanks, again.

closed 2022-01-25 by TricksterSam