system76/ec #273: feature request: move ectool to stable Rust

anthonyjmartinez, 2022-03-09:
It appears at this moment that the nightly requirement is due to the dependency on `redox_hwio` where there appear to be two nightly features in use:
- [ ]  `asm!`, stabilized in Rust 1.59.0 (and I believe referenced [here](https://gitlab.redox-os.org/redox-os/hwio/-/merge_requests/4))
- [ ] `const_fn_trait_bound`, which appears to be tracking stabilization for Rust 1.61.0 ([ref](https://github.com/rust-lang/rust/pull/93827))

If the above is correct is it possible to trend towards use of stable Rust? Thanks!

anthonyjmartinez, 2022-03-09:
Thanks, the referenced here link in that item is to that very MR.

I assume to try the assembly macro from stable it would be necessary to switch to a nightly branch after 1.59.0 was released, and then use the feature from `std::arch::asm`?

crawfxrd, 2022-03-09:
redox\_hwio has a `stable` feature gate, so I need to update all the points where `asm` is used.

But is there any reason you want the `redox_hwio` feature for ectool on stable? It's only used by `firmware-update`, which won't be on stable either.

You can build on stable right now with:

```
cargo +stable build --release --manifest-path tool/Cargo.toml
```

anthonyjmartinez, 2022-03-12:
In general I'd prefer all production Rust be stable, but I don't have a specific rhyme or reason for this beyond a hesitance to depend on that which might never stabilize.

I think I can do what I need to do from the basis of prior art and stable Rust without needing the entire ectool, so I will approach it that way. This can probably be closed.

closed 2022-03-12 by crawfxrd