system76/ec #280: gaze17: Power button LED stays on when system is suspended

linuxgnuru, 2022-06-13:
### Expected behavior

When the laptop suspends the power button LED either blinks or turns off

### Actual behavior

Power button LED remains on when laptop suspends

crawfxrd, 2022-06-13:
Are any external devices plugged in?

Is the CSME enabled? A quick check is to see if device 16.0 is present in `lspci`.

linuxgnuru, 2022-06-13:
No external devices plugged in.  Attaching output from `lspci

[lspci.txt](https://github.com/system76/ec/files/8894869/lspci.txt)
`

crawfxrd, 2022-06-14:
```
root@gaze17:~# cat /sys/devices/system/cpu/cpuidle/low_power_idle_cpu_residency_us
11425730
root@gaze17:~# cat /sys/devices/system/cpu/cpuidle/low_power_idle_system_residency_us
0
root@gaze17:~# cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec
0
```

CPU is reaching C10. System is not reaching SLP_S0.