system76/ec #282: Oryp7: Fn+PrtSc does not perform SysRq

MilesBHuff, 2022-06-15:
The keyboard on the oryp7 says that Fn+PrtSc should perform SysRq, but this does not happen;  instead, Fn is ignored and the computer receives a PrtSc keycode.

jackpot51, 2022-06-16:
SysRq is actually the same keypress as PrtSc. Use Alt+PrtSc to use magic sysrq.

MilesBHuff, 2022-06-16:
Oh, it's Alt+PrtSc, not Fn+PrtSc?

jackpot51, 2022-06-16:
Run `sudo dmesg -w` in a terminal

Press Alt+PrtSc+Space all at the same time, on the internal laptop keyboard

Magic SysRq options will be printed in the terminal. Tested on my own oryp7.

MilesBHuff, 2022-06-16:
Ah!  So it's Alt+PrtSc+Space.  Is this documented anywhere?
One would think, per the key, that it would just be Fn+PrtSc.
Why the complex combination?

jackpot51, 2022-06-16:
Can I assume you have it working now? If so, please close this issue.

jackpot51, 2022-06-16:
If you want more detail, you can see it here: https://www.win.tue.nl/~aeb/linux/kbd/scancodes-1.html

On PS/2 style keyboards like most internal laptop keyboards, PrtSc and SysRq send the same scancode, and Pause and Break send the same scancode. The OS uses additional modifiers on those like Shift, Ctrl, or Alt to determine the action. Definitely not how I would have done it, but it is how it is and is not a firmware related issue.

MilesBHuff, 2022-06-16:
Thank you for the detailed explanations, @jackpot51!