system76/ec #283: Oryp7: Fn+F7 does not toggle the display

MilesBHuff, 2022-06-15:
Fn+F7 is labelled as toggling displays;  but what actually happens when I press it, is Semicolon on keydown, and Super_L on keyup.  This is obviously incorrect.

leviport, 2022-06-15:
Did you connect an external display? That hotkey only works with an external display connected. It looks like it's working fine on our oryp7.

MilesBHuff, 2022-06-15:
Yes.
I'm looking at the output of `xev`, fwiw.
I'm using Pop!_OS 22.04 and the latest EC code from GitHub.

MilesBHuff, 2022-06-15:
Regardless of whether it's working for me with an external display, the key combo should definitely not be sending Semicolon and L_Super.  If a monitor is disconnected, Fn+F7 should either do nothing, or pass through to a plain F7.

jackpot51, 2022-06-16:
What keyboard layout are you using? `Fn+F7` will send `Super+P` which may be translated differently if you don't have US QWERTY selected.

MilesBHuff, 2022-06-16:
Ah!  That could be it:  I'm using Colemak.  Let me try QWERTY.

MilesBHuff, 2022-06-16:
Is there a way to make a keyboard-agnostic version of this in the EC?

jackpot51, 2022-06-20:
There is no way to detect the operating system keyboard layout from the EC, so no.

jackpot51, 2022-06-20:
I recommend to just use Super+P in your own keyboard layout instead of the Fn+F7 hotkey.