system76/ec #331: Oryx8 - Laptop suddenly dies if stressed on battery

rnd-ash, 2023-01-14:
Updated to latest EC and Coreboot version via system76-firmware-cli , and even tried the latest EC git code and have the same results.

Side note. I am using optimus-manager here to disable the DGPU on battery, will post an update comment if enabling the DGPU on battery fixes this to narrow down the issue.

When on battery power after a fresh boot, the laptop works fine, I can compile code and stress the system without issue. But simply after sleeping the notebook then resuming it, the laptop suddenly dies if anything moderately stressful is executed (Even launching chrome causes it!)

Once dead, the power button does nothing for around 20-30 seconds, then it responds and lets the notebook power up again.

rnd-ash, 2023-01-14:
I enabled the DGPU on battery (Switched from optimus-manager to system76-power and used hybrid graphics mode), and get the same result. If anything the laptop might be a bit more sensitive after sleeping (Lighter actions cause it to die), so maybe its a power budget issue after the system has resumed from sleep?

rnd-ash, 2023-01-15:
Another update. If I resume the notebook with the AC adapter plugged in, then unplug the adapter once the notebook has woken up, the issue does not occur when stressing it.

crawfxrd, 2023-01-20:
Able to reproduce on firmware 2022-11-21_b337ac6, which is what everything other model is using for the release.

crawfxrd, 2023-01-30:
Introduced by #298.

rnd-ash, 2023-02-02:
> Able to reproduce on:
> 
> * 2022-11-21_b337ac6, which is what everything other model is using for the release
> * EC `master` with coreboot 4.19 rebase
> 
> Does not happen with:
> 
> * coreboot 4.19 and [d3ee037](https://github.com/system76/ec/commit/d3ee0378792c4afbe069d6d8ed8c02f2eef2a0cc) (commit used for the 2021-09-23_b9b0e89 release)
> * coreboot 4.19 and [20d40f3](https://github.com/system76/ec/commit/20d40f347741880ee27757bf368691b82cc60bfa)
> 
> Based on the sound the fans make, I think the power limits are lost after S3. With [d3ee037](https://github.com/system76/ec/commit/d3ee0378792c4afbe069d6d8ed8c02f2eef2a0cc) they don't spin up while building the kernel (implying the 28W limit is still set), whereas on `master` they spin up pretty soon in the build process.

Can confirm. With i7z open, even with governer set to powersave, I see the CPU jump to 4100Mhz just before the notebook dies.

Also, it appears restarting the notebook after resuming is not a fix for this. It has to be shut down, left off for a few seconds, then started for it to fix itself.

crawfxrd, 2023-02-24:
@jackpot51 has proposed this change:

```diff
diff --git a/src/board/system76/common/power.c b/src/board/system76/common/power.c
index 93189c57d19b..bc7d928262a3 100644
--- a/src/board/system76/common/power.c
+++ b/src/board/system76/common/power.c
@@ -354,6 +354,8 @@ void power_cpu_reset(void) {
     fan_reset();
     //TODO: reset KBC and touchpad states
     kbled_reset();
+
+    power_set_limit();
 }
 
 static bool power_button_disabled(void) {
```

This fixes the issue on my oryp8.

rnd-ash, 2023-03-02:
I tried this fix that was suggested. 

Granted, the fans don't spin up now, and web browsing works most of the time, but I am still encountering the laptop just shutting off when on battery after a resume.

Maybe there are other devices that need their power states reconfigured when the system awakens?

rnd-ash, 2023-04-03:
Hey @crawfxrd ,

I managed to capture it on video. Seems opening Chrome after resuming the laptop (Even with the fix) is guaranteed to make the laptop die:

https://photos.app.goo.gl/hgrcq83iNJYEFk2p6

Listen to the video though, its instant, there is no fan spin prior the the laptop cutting out, I just clicked on the Chrome icon and instantly the system died.

rnd-ash, 2023-04-12:
I've flashed 930f16b230a1574abddc1cf313c32f1a90c994e9, and good news is so far I cannot replicate this! 

I'll run the laptop for a few more days on battery and see if it at all pops up again, otherwise I will close this issue.

Sidenote worth mentioning, don't know if it is a consequence of it working now, but on battery power after a resume, animations seem sluggish all over the desktop. I ran some stress tests and noticed that the CPU was never going past 1.4Ghz. Even when loading just 1 or 2 cores.

closed 2023-06-16 by crawfxrd