system76/ec #343: Keyboard backlight color changes to white after suspend & resume

leviport, 2023-03-16:
On oryp7 and gaze18, I noticed that suspending and resuming changes the keyboard backlight color to white. However, hitting Fn + numpad / (color change hotkey) changes to the next color in the rotation, following the color that was set before suspending. For example, if I change the color to Pink, hitting Fn + / again changes the backlight to Green, which is the color that always comes after pink while cycling through keyboard colors. So it seems like something knows what the color _should_ be, but it's still changing to white after a suspend and resume cycle.

n3m0-22, 2023-03-16:
I can confirm this is also happening on the `gaze15`, but not the `oryp4`.

leviport, 2023-03-16:
Oryp4 has proprietary firmware and EC, so that makes sense. Is oryp4's keyboard the same color after suspending and resuming?

n3m0-22, 2023-03-16:
Yes the color is the same after suspend/resume.

uSpike, 2023-03-17:
This happens on my `darp8` as well.

leviport, 2023-03-17:
My darp5 with open firmware also does this. As I suspected, bonw14 does not; the keyboard comes back the same color after resuming.

crawfxrd, 2023-03-17:
Caused by `kbled_reset` being called in `power_cpu_reset`, which happens on S3.

https://github.com/system76/ec/blob/1046dd7aec37409d0c077ea925856bc3c2a70b81/src/board/system76/common/power.c#L347-L357

crawfxrd, 2023-03-24:
If I unload system76-acpi, the keyboard turns on with the same color it suspended with, before turning off again and then being white.

crawfxrd, 2023-03-27:
KB backlight is also being [reset by ACPI `WAK`](https://github.com/system76/coreboot/blob/fbccafec553f8382403aa6ec36b57ba0e35458c0/src/ec/system76/ec/acpi/ec.asl#L94-L95).