system76/ec #347: I need help!

2805852115, 2023-04-28:
I don't know how to modify the parallel. c file. Has Guys sent me a modified one? Thank you very much
I am using a 1mm FPC cable

jacobgkau, 2023-04-28:
> I don't know how to modify the parallel. c file. Has Guys sent me a modified one?

What file are you referring to, and who is "Guys"?

---

System76 customers can reach out to [support](https://support.system76.com/) for technical assistance. For non-System76 hardware, you can seek community support on [Reddit](https://reddit.com/r/pop_os/) or [Mattermost](https://chat.pop-os.org).

jacobgkau, 2023-04-28:
If you're talking about the pins in https://github.com/system76/ec/blob/master/src/board/arduino/mega2560/parallel.c, you'd need to say which model of computer you have for someone to give you the proper pin mapping.

2805852115, 2023-04-28:
> If you're talking about the pins in https://github.com/system76/ec/blob/master/src/board/arduino/mega2560/parallel.c, you'd need to say which model of computer you have for someone to give you the proper pin mapping.

Sorry, because my computer is broken and I'm in a hurry,I have revised my issue

closed 2023-06-01 by crawfxrd