system76/ec #350: change FN-key behavior

Vitamin-X, 2023-05-10:
Hey, i got a question 

is it possible to change FN+F2 (which normaly turns off the display) to change the Advanced Nvidia Optimus Mux state?.... its nearly impossible to get it working on linux and i honnestly just need to switch outputs without everything else it just needs to turn GPI2 from off to on

notebook codename: PC5x_7xHP_HR_HS
EC: IT5570

jackpot51, 2023-05-10:
Are you asking about a specific System76 model?

Vitamin-X, 2023-05-10:
i got an "Clevo PC70HR" and it seems like it got the same board then the oryp8 (so popably?) but i gues its also usefull for other linux user with Advanced Optimus Notebooks

i am atleast trying to use the GPU in a VM and wanna switch the output to the VM so i dont need to use the nvidia stuff in linux at all

Vitamin-X, 2023-05-10:
worth another comment..

tryed to understand the files for a while and i gues i got the functionality for the mux on the F7 key working, but can someone check if i didnt made any mistakes?

https://github.com/system76/ec/compare/master...Vitamin-X:ec:master

crawfxrd, 2023-05-10:
Base unit of oryp8 is the PC[57]0HS.

You will fist need to make it compile, which requires you at least:

- Define `K_MUX_TOGGLE` somewhere
- Add `K_MUX_TOGGLE` to `key_should_repeat()` so it returns `false`
- Add a guard so only boards that define `MUX_CTRL_BIOS` can use it

But I'm unsure what you're trying to do:

> i am atleast trying to use the GPU in a VM and wanna switch the output to the VM so i dont need to use the nvidia stuff in linux at all

Do you want PCI passthrough, so that the host is using the iGPU and the VM is using the dGPU? (Not that I know how to set that up either.)

Vitamin-X, 2023-05-11:
yea the VM is already running fine and i can see it on the external display, but i wanna switch the internal display to the dGPU and would love to do it by the push of an button.

> PC[57]0HS

the HS just means its a diffrent GPU mine got an 3070 and i gues the HS was the 3060 there is also an HP with an 3080

> * Define `K_MUX_TOGGLE` somewhere
 
ah yea i forget about that, now i just renamed the old key to the new "name"

> * Add `K_MUX_TOGGLE` to `key_should_repeat()` so it returns `false`

and thats so you dont need to release Fn to push it again right?

> * Add a guard so only boards that define `MUX_CTRL_BIOS` can use it

how can i do that? and is it realy needed?

Vitamin-X, 2023-05-11:
ah our german keyboard seems to get one key more right next to the left shift with "<>|" and no clue which pins they use in the matrix, i hope thats not messing all up

`root@timo-PC70HR:~/ec# make BOARD=system76/oryp8 KEYMAP=german
  CC        system76/oryp8/2023-05-11_4d8bb55/scratch/board/system76/common/scratch/../smfi.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/scratch/board/system76/common/scratch/main.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/scratch/board/system76/common/scratch/stdio.rel
  LINK      system76/oryp8/2023-05-11_4d8bb55/scratch/scratch.ihx
  OBJCOPY   system76/oryp8/2023-05-11_4d8bb55/scratch/scratch.rom
  XXD       system76/oryp8/2023-05-11_4d8bb55/include/scratch.h
  CC        system76/oryp8/2023-05-11_4d8bb55/flash/board/system76/common/flash/main.rel
  LINK      system76/oryp8/2023-05-11_4d8bb55/flash/flash.ihx
  OBJCOPY   system76/oryp8/2023-05-11_4d8bb55/flash/flash.rom
  XXD       system76/oryp8/2023-05-11_4d8bb55/include/flash.h
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/main.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/arch/8051/arch.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/arch/8051/delay.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/arch/8051/time.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/acpi.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/battery.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/charger/bq24780s.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/config.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/dgpu.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/ecpm.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/espi.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/fan.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/flash/wrapper.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/gctrl.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/kbc.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/kbled.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/kbled/rgb_pwm.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/kbscan.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/keymap.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/lid.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/parallel.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/peci.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/pmc.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/pnp.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/power.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/ps2.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/pwm.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/scratch.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/smbus.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/smfi.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/stdio.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/usbpd/none.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/common/wireless.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/oryp8/board.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/board/system76/oryp8/gpio.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/common/i2c.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/common/keymap.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/common/version.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/ec.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/espi.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/gpio.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/i2c.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/kbc.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/pmc.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/ps2.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/ec/ite/signature.rel
  CC        system76/oryp8/2023-05-11_4d8bb55/keyboard/system76/15in_102/keymap/german.rel
  LINK      system76/oryp8/2023-05-11_4d8bb55/ec.ihx
  OBJCOPY   system76/oryp8/2023-05-11_4d8bb55/ec.rom
Built 2023-05-11_4d8bb55 for system76/oryp8
`

but so far no errors, now i need to be brave enouth to test it i gues

Vitamin-X, 2023-05-11:
i think i got all mistakes fixed now.. and made it optional with "CONFIG_MUX_COMBO=y" so its still compatible to the other models, it also seems like that FN+F7 is just a replacement for super+P soo you might like to merge that key combination to your code too cuz the default option is not realy an shortcut and switching the mux is soooooo much more usefull

[image] 

the picture on the key also pretty much matches that behavior

closed 2023-06-21 by crawfxrd