system76/ec #374: ectool console not working

curiousercreative, 2023-07-11:
I've tried running this on master and from the commit of my device's latest firmware release with no joy. I used to get a ton of output when running this and it was how I'd debug EC firmware I was writing. Any guidance is appreciated.

```
$ git pull origin master
$ ./scripts/deps.sh
$ ./scripts/ectool.sh info
warning: unused manifest key: registries
warning: use of deprecated unit variant `hidapi::HidError::OpenHidDeviceError`
   --> src/main.rs:431:43
    |
431 |                     Err(hidapi::HidError::OpenHidDeviceError.into())
    |                                           ^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: `system76_ectool` (bin "system76_ectool") generated 1 warning
    Finished release [optimized] target(s) in 0.01s
board: system76/galp5
version: 2023-06-12_53e9ac6
```

```
$ ./scripts/ectool.sh console
warning: unused manifest key: registries
warning: use of deprecated unit variant `hidapi::HidError::OpenHidDeviceError`
   --> src/main.rs:431:43
    |
431 |                     Err(hidapi::HidError::OpenHidDeviceError.into())
    |                                           ^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: `system76_ectool` (bin "system76_ectool") generated 1 warning
    Finished release [optimized] target(s) in 0.01s
```

crawfxrd, 2023-07-29:
I have used `make console_internal` to test kbscan without issue on the latest code.

curiousercreative, 2023-07-31:
@crawfxrd aha, I indeed got console working with that make command. I might open a PR to add this to our debugging documentation. Do you know how I can get these [TRACEs](https://github.com/system76/ec/blob/master/src/board/system76/common/fan.c#L73) to show in this console?

crawfxrd, 2023-07-31:
Debug level is set at compile time. You would need to change the level in `src/board/system76/common/common.mk`, or change those to a higher level (to avoid spew from other `TRACE` lines).