system76/ec #381: "peci_get_temp: hardware error: 0x0A" on lemp12

codewiz, 2023-07-19:
On a new lemp12 running the stock 2023-05-16 firmware, the CPU fan is constantly blowing at half speed (128). This happens at boot time as well as when running Pop!_OS or Fedora Rawhide.

```
giskard:~# system76_ectool info
board: system76/lemp12
version: 2023-05-16_e9b9ea8
giskard:~# system76_ectool fan 0
128
```

Some things I tried:

- Pressing Fn + 1 causes the fan to spin at maximum speed (255), and pressing it again returns to "automatic", which is locked to 128.

- Setting the power profile to "Power Save" in KDE, or via `system76-power profile` does not change the speed.

- Setting the minimum fan speed (`system76_ectool fan 0 0`) or the maximum speed (`system76_ectool fan 0 255`) has a noticeable effect, but the EC immediately brings the speed back to 128.

It looks like this could be the root cause:

```
bernie@giskard:~/src/system76/firmware-open/ec% sudo system76_ectool console
peci_get_temp: hardware error: 0x0A
peci_get_temp: hardware error: 0x0A
peci_get_temp: hardware error: 0x0A
...
```

Originating from: https://github.com/system76/ec/blob/64e5b8308db38a03822e97e43fc9118c82433f76/src/board/system76/common/peci.c#L324

crawfxrd, 2023-07-21:
0xA is `WR_FCS_ERR | FINISH` (FCS: Frame Check Sequence).

codewiz, 2023-07-23:
> 0xA is `WR_FCS_ERR | FINISH` (FCS: Frame Check Sequence).

Does this occur on all lemp12 units running the same firmware, or is this a symptom of a faulty serial interface with the PECI?

crawfxrd, 2023-07-24:
Lab unit doesn't have the issue. No other boards I have are experiencing this issue. So may be an issue with the board.

codewiz, 2023-08-08:
Looks like there was a hardware issue with my unit. I returned it to System76 and received a replacement.

Closing, since it's not a firmware bug.

closed 2023-08-08 by codewiz